adafruit/DHT-sensor-library #30: `return` from inside `switch`-`case`

Mausy5043, 2015-02-16:
I noticed in the functions `readTemperature()` and `readHumidity()` that the result is returned from inside the case using `return f;`. So in one switch-case construct there are multiple `return`s. One for each case.

I was wondering if this is by design or not. I expected to see something like:
```
switch {
  case A:
    f = foo;
  case B:
    f = bar;
}
return f;
```

Is there some advantage to choose the current construct over the other?

Even more confusion is http://arduino.cc/en/Reference/SwitchCase where each `case` is ended in a `break;`. Won't that happen automagically when the appropriate `case`-code has executed?

dhammika-marasinghe, 2015-03-31:
if we are returning value then we can't use break statement, but for other cases we need to add break statement to exits the switch.

Mausy5043, 2015-03-31:
I tend to disagree.

Mausy5043, 2015-03-31:
See PR #32 for what I mean.

dhammika-marasinghe, 2015-03-31:
noted the problem. Thanks +1

Mausy5043, 2015-03-31:
You're welcome. Have fun! :+1:

Mausy5043, 2015-03-31:
I'm keeping this pull request open because it provides the functions with a single exit point which is IMHO more easily readable/understandable for beginners.
Up to @ladyada to decide completing the pul or not.

dhammika-marasinghe, 2015-03-31:
yes it is. And you code looks stable. I think @ladyada will consider that.

Mausy5043, 2015-04-05:
PR #33 should resolve this issue. Closing.

closed 2015-04-05 by Mausy5043