adafruit/DHT-sensor-library #34: DHT-Library does not work with esp8266/Arduino 1.6.4

Thro42, 2015-05-14:
I use this library in es ESP8266 projekt. After I updated the Arduino IDE to 1.6.4 and use the bord installation of toddtreece (https://github.com/toddtreece/esp8266-Arduino/releases), i got runtime errors.
Fatal exception (28): 
epc1=0x40241d6f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000084, depc=0x00000000
A saple Coding is
#include \<DHT.h\>
void setup() {
  Serial.begin(115200);
  Serial.println("Setup... ");
}
void loop() {
}

tdicola, 2015-06-26:
That's an odd error, it loks like something with the Arduino IDE is failing.  I've actually been using the DHT sensor code with the ESP8266 recently and didn't run into any similar issues so I have feeling it could be something with how the ESP8266 arduino core is installed.  Try deleting Arduino's cache folder to clear out any cores the board manager downloaded (but just as a warning this will remove any custom cores you've installed with the board manager--your sketches and libraries won't be touched and should be fine though).  You can find the cache folder on Windows in C:\Users\<your username>\AppData\Roaming\Arduino15 (the AppData folder is usually hidden), or on Mac in ~/Library/Arduino15/, or on Linux in ~/.arduino15/.  Completely delete that folder and then start Arduino again to see the board manager is clear of any custom boards.

Once the board manager is clear then you can install the ESP8266 arduino core again by following their official install steps here: https://github.com/esp8266/Arduino  Open up preferences in the Arduino IDE and add the following board manager URL that the ESP8266 Arduino folks publish: http://arduino.esp8266.com/package_esp8266com_index.json  Then open the board manager and install the ESP8266 core again.

Once you've got the core reinstalled one thing to note when using a DHT sensor with the ESP8266 is that you need to slightly adjust some timing values so the ESP8266's faster processor can read the data from the sensor.  The ESP8266 Arduino github page actually mentions what to do:
```
DHT11 - initialize DHT as follows: DHT dht(DHTPIN, DHTTYPE, 15);
```
So when you use an example DHT application make sure to add that extra 15 parameter to the DHT object's constructor call.  The parameter tells the code to wait an extra amount of cycles to read ones and zeros from the sensor--normally the library assumes its running on a slower Arduino so it can get a little confused when the processor is faster.  In my testing the 15 cycle count works great with all the DHT sensors on the ESP8266.

closed 2015-06-26 by tdicola

asimkumar2222, 2015-12-19:
i have a similar problem, that was caused by insufficient current from power supply, it consumes around 750mA peak current.

ckwiek, 2016-08-27:
Hello tdicola,

I had a similar problem, no sketch worked with its library, which had always worked, I tested my applications with ESP8266 and did not work with Arduino yes they worked.

Searching, I found your comment here, and did the procedure to delete the Arduino 15 folder, but still had no success, until I found the cause of my problem.

At some time, I met the site Iot iotplaygrond, a very good site and I really liked, as ThingSpeak taste where I have several applications, I realized that the applications of iotplaygroud gift DHT library functioned agreement, but the other with ESP8266 not then discovered the problem, I downloaded a library in order to use the iotplaygrond applications, which is this: https://github.com/iot-playground, but his did not work and had always error, then only let your in librirys and everything worked normal folder, compiles without problems or errors, but now scketchs of iotplygroud not compile to solve my problem provisionally, I'm doing the following, put your library whenever the folder, and when I have to compile sketches of your retreat iotplayground and put the other.

Do you think there should be some solution to not have to act this way?

Thanks in advance.

Regards,

Carlos Kwiek