adafruit/DHT-sensor-library #37: Release 1.1.0 produces lots of NANs

sandymac, 2015-07-18:
FYI: Release 1.0.0 works well for me, but when I switched to release 1.1.0 I get nans about 50% of the time.

tdicola, 2015-07-20:
Ah interesting, just to check what processor/board are you both using with the sensor?

sandymac, 2015-07-21:
My board is Arduino UNO R3 made by Arduino.org and my sensor is a Aosong AM2302.

alewir, 2015-07-21:
Also UNO R3 with DHT22.

desinho, 2015-07-21:
Same issue, Arduino Mini Pro with AM2302

tdicola, 2015-07-22:
Thanks for all the details!  I took a closer look with an AM2302 and also saw the intermittent failures.  Digging in a little deeper it looks like the problem was with digitalRead being a little too slow to reliably read the sensor's pulses (which are ~20-80 microseconds long).  I switched over the logic to use direct GPIO port access on AVR and it seems rock solid now with no NaN results.  Grab the 1.1.1 version or github head and let me know if you run into any issues, thanks!

closed 2015-07-22 by tdicola

matthijskooijman, 2015-07-24:
@tdicola Did you consider using Arduino's `pulseIn()` function directly? That one should actually already be optimized for whatever platform you're running on. On Arduino 1.6.5 it even uses an assembly loop for fully known cyclecounts and maximum accuracy, but earlier versions are already accurate enough I believe (though < 1.6.0 need a specific gcc version).

Using pulseIn could greatly simplify this library's code and memory usage again (the current code needs 320bytes of stack, that's likely to cause problems when memory is low...). For reference, here's an implementation: https://github.com/matthijskooijman/DHT-sensor-library/blob/pulsein/DHT.cpp#L118-L147

It's still based on the library before all the recent refactoring, I should have fetched before I started coding. Still, if you think this looks promising I could rebase it on top of the current code?

tdicola, 2015-07-28:
Oh nice, yeah I just had noticed the changes to pulseIn and wasn't sure if it would work on other platforms.  However if looks like you've got it all covered with code for AVR, Due, etc--awesome!  Yeah feel free to send a pull request that refactors the library to use pulseIn.  Thanks!

commit 85d85170 (2015-07-22): Fix #37 by optimizing pulse read to use direct port access on AVR, stops checksum failures from missed signals.