adafruit/DHT-sensor-library #40: Works with some ESP03's and not with others

raimohanska, 2015-08-18:
This is probably not an issue with the library, but I've noticed that out of 3 esp-03 modules I've hooked up to my breadboard, only one is able to pick a temp/humidity reading from a DHT-22 sensor. The boards are from the same batch I ordered, and are wired up similarly.

I'm using the exact same breadboard setup with all the modules. I've verified that the GPIO2 port that I'm using is working on all of the modules, i.e. I can read and write using the port no problem. Still, reading from DHT22 fails using the same port.

Any ideas what could be wrong?

franc-carter, 2015-08-18:
I'be been having problems with the DHT22 on esp-01

I've been experimenting today and I'm beginning to suspect power, the experiments are early but evidence is mounting

 The NodeMCU works fine hooked up on my desk and I have an esp-01 running on my desk now, it's much more stable than the the ones I have on longer power cables around my apartment.

raimohanska, 2015-08-18:
Oh, I switched from my 3.3 regulator setup to 2*AA batteries and I get readings now. Interesting.

franc-carter, 2015-08-18:
Another experiment pointing in the same direction - thanks. I have found
that the esp chips are really sensitive to power that isn't clean.

Do you know if AA batteries are going to better at providing a steady
current ?

raimohanska, 2015-08-18:
I'm afraid the AA batteries won't work very well, as their voltage is bound to sink from ~3.3 to 3 volts gradually.

franc-carter, 2015-08-18:
Yeah good point, but at least it points to the problem being power

raimohanska, 2015-08-18:
FYI, I'm using a MCP1703 3-pin SOT23-A regulator with two 1uF electrolytic capacitors. This setup works fine with 1 of my 3 ESP-03 boards, but the temp/humidity measurement fails with the other two boards. Pretty strange from my point of view. I've tested with several DHT-22 units and the result is the same.

franc-carter, 2015-09-01:
I suspect that it's related to RF interference ;-(

I moved my DHT22 20cm away from the ESP-01 and the and the readings went from highly erratic to reasonably stable.

tdicola, 2015-10-14:
Ah interesting info that RF and power can destabilize the readings.  If you can get a scope or logic analyzer on the DHT output pin it might help check that you're getting a good clean signal or if there's noise.  The signal isn't super fast (~20 microsecond pulses at the fastest) so even an Arduino Uno running as a logic analyzer should be able to capture it: http://forum.arduino.cc/index.php?topic=52881.0  Good luck!

closed 2015-10-14 by tdicola