adafruit/DHT-sensor-library #44: Problem during compilation in Arduino DUE board

CharlyBote, 2015-10-10:
I receive this error when compile the test sketch.
C:\Users\charly\Documents\Arduino\libraries\DHT\DHT.cpp -o C:\Users\charly\AppData\Local\Temp\build5143615002772527901.tmp\DHT sensor library\DHT.cpp.o 
C:\Users\charly\Documents\Arduino\libraries\DHT\DHT.cpp: In constructor 'DHT::DHT(uint8_t, uint8_t, uint8_t)':
C:\Users\charly\Documents\Arduino\libraries\DHT\DHT.cpp:15:9: error: invalid conversion from 'Pio*' to 'uint8_t {aka unsigned char}' [-fpermissive]
   _port = digitalPinToPort(pin);
         ^
Error de compilación

Version of arduino 1.6.5
Board: DUE
Sensor: DHT22

tdicola, 2015-10-13:
Oh ack, thanks for catching the issue!  Just took a look and realized it was a problem with the 8-bit AVR specific GPIO access code.  I put this behind an #ifdef that keeps it from being a problem on platforms like the Due.  Also gave it a quick test with a Due and it works great now.  Grab the latest version of the code (version 1.2.3) and it should be resolved, thanks!

closed 2015-10-13 by tdicola

CharlyBote, 2015-10-18:
Thank you very much to respond.
I will probe this today.

Thanks again

best regards.

-- 
Carlos Botejara
IT Analyst
cbotejara@gmail.com
NEUQUEN - ARGENTINA
C: 0299-154230346
MSN:carlos.botejara@hotmail.com <carlos.botejara@hotmail.com>
*SKYPE: carlos.botejara*
LINKEDIN: http://www.linkedin.com/in/carlosbotejara

Este correo está dirigido únicamente a la persona o entidad que figura en
el destinatario y puede contener información confidencial y/o privilegiada.
La copia, reenvío, o distribución de este mensaje por personas o entidades
diferentes al destinatario está prohibido.
Si Ud. ha recibido este correo por error, por favor contáctese con el
remitente inmediatamente y borre el material de cualquier computadora.
Este correo puede estar siendo monitoreado en cumplimiento de esta política.

commit 15020aa8 (2015-10-13): Fix #44 by conditionally excluding unused port and bitmask state on non-AVR platforms.