adafruit/DHT-sensor-library #49: DHTServer:65: error: 'gettemperature' was not declared in this scope

mikejclark, 2015-12-05:
Not able to compile this code taken from the tutorial
https://learn.adafruit.com/esp8266-temperature-slash-humidity-webserver/code
Am I doing something wrong?
Thanks

Arduino: 1.6.6 (Windows 10), Board: "Generic ESP8266 Module, Serial, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck"

C:\Users\mike\Documents\Arduino\DHTServer\DHTServer.ino: In lambda function:

DHTServer:65: error: 'gettemperature' was not declared in this scope

     gettemperature();       // read sensor

                    ^

C:\Users\mike\Documents\Arduino\DHTServer\DHTServer.ino: In lambda function:

DHTServer:71: error: 'gettemperature' was not declared in this scope

     gettemperature();           // read sensor

                    ^

exit status 1
'gettemperature' was not declared in this scope

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

asetyde, 2015-12-05:
you create object DHT dht ?

dht.readTemperature();

Please if you use espmodule , if you note some random block of board after 2 min 1h or more hours please write us

mikejclark, 2015-12-05:
Hi. I'm new to all things esp and arduino so trying my best to follow the tutorial as closely as possible. There is no mention of creating an object? My code is the same as the tutorial, except for my WiFi credentials

mikejclark, 2015-12-05:
Found this:
http://iot-playground.com/2-uncategorised/67-arduino-esp8266-ide
Mentions there is a problem with 1.6.6.
Complied with no errors in 1.6.5

closed 2015-12-05 by mikejclark

asetyde, 2015-12-05:
my code randomly block on stable 2.0.0 , all work on 1.6.5

knnniggett, 2015-12-22:
For those that prefer to build using the latest version of Arduino IDE (currently 1.6.7), simply move the block of code that represents the gettemperature() call up, prior to the definition of setup(void). It builds fine once you make that change.