adafruit/DHT-sensor-library #50: DHT with esp8266/arduino 2.0.0 stable random blocks

asetyde, 2015-12-06:
Hi , i've found some strange issue use dht22 library with esp8266/arduino git stable 2.0.0 , random blocks of device . 

I don't understand why but i think about memory leak

W-E, 2015-12-08:
I had this lib unmodofied running for days without lockups on an ESP-12, also programmed with esp2866/Arduino without any lockups.
The lib is not big, you can read it in 1/2 hour. I do not see memory leaks in there.
Check the hardware: clean levels at the I/O pins, stable supply. Many strange issues result from unstable / insufficient supply. ESP8266 has high current consumption spikes, insufficient power supply stability ( too small capacitors or too high ESR ones ) causes issues.

asetyde, 2015-12-08:
I use nodemcu v1.0 ... You ?

W-E, 2015-12-09:
I do program the ESP8266 in C via the Arduino IDE and the arduino standard
libs ported to the ESP8266.
https://github.com/esp8266/Arduino
But that does not matter. The DHT lib keeps te data line held up when the
DHT tries to pull it down. That's all. That is not related to the used
toolchain.

> I use nodemcu v1.0 ... You ?
>
> Sent from my iPhone

asetyde, 2015-12-09:
Ok , about your expirience ,4.7  resistance is useful ?  I not use with my nodemcu , use integrated resistance pull up

osos, 2016-02-11:
See: https://github.com/esp8266/Arduino/issues/654 and https://github.com/esp8266/Arduino/issues/1594

Seems the new SDK for esp8266 does not allow Interrupts to be disabled.

Interrupt-driven DHT lib can be found here: https://github.com/chaeplin/PietteTech_DHT-8266

@igrr  suggest to use AutoInterrupt(1) - as this will only block level 1 interrupts and not all interrupts.

Makuna, 2016-02-11:
While switching to autointerruptlock will be a good thing here, the code that is timing critical is NOT guaranteed to function correctly.  With the 50us timing per pulse it is unlikely to be an issue though as the nmi interrupt tend to below 8us.

The loop in expectpulse states it will exit if it takes longer than  1ms.  That does not seem to be the case.  The code in the loop isn't checking clock cycles, it's checking loop counts, but the max cycle was based off clock cycles.  You can compare clock cycles instead of loop counts by using esp.getcycles ().  Not sure what the avr version was thinking they were comparing unless someone look at the assembly code to confirm those statements were only a cycle.

One last bit of advice, put timing critical code in a c file and mark ithe function as ram cache, without this, initial pulse timing can get way off as the routine is moved from flash to ram to be run, as the code seems to stutter until it's cached.

Ghostbird, 2016-06-25:
DHT11  on NodeMCU 0.9 with this library always fails to read. No reason found yet.
Tested the sensor by plugging it into an Arduino Nano and it worked right away.

closed 2016-10-28 by asetyde