adafruit/DHT-sensor-library #53: Error: 'microsecondsToClockCycles' was not declared in this scope

ramonmata, 2015-12-29:
Hello All,

I am working with the Arduino IDE 1.6.4 which comes with the Intel Edison Dev Software, and I am getting this error.

Can you tell where can I find any documentation for 'microsecondsToClockCycles'?  Can't find any on the web

regards

ToddKlindt, 2016-01-17:
I'm getting the same error. I can't get the example to compile. I'm using the 1.6.7 IDE with a Linkit One.

tk

rplankenhorn, 2016-02-14:
Any update on this?  I am seeing the same issue with an Intel Edison.

SompopYong, 2016-03-16:
Hi,

I'm using a LinkIt ONE and found this problem when try to use DHT22 sensor.
The error show that 
`C:\Program Files (x86)\Arduino\libraries\DHT\DHT.cpp:23:46: error: 'microsecondsToClockCycles' was not declared in this scope

   _maxcycles = microsecondsToClockCycles(1000);  // 1 millisecond timeout for`

I found that a constant "microsecondsToClockCycles" is define in "C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino\Arduino.h".

So, I copy the define statement below and add to "DHT.cpp" 
`#define clockCyclesPerMicrosecond() ( 260L ) //260 is Clock Cycle of LinkIt ONE in MHz
#define clockCyclesToMicroseconds(a) ( (a) / clockCyclesPerMicrosecond() )
#define microsecondsToClockCycles(a) ( (a) * clockCyclesPerMicrosecond() )`

The verify is pass, and can be upload and run correctly.

Hope this help you.

mikhailrojo, 2016-07-20:
wow SompopYong you are so cool!
thanks

closed 2018-08-09 by ramonmata