adafruit/DHT-sensor-library #57: nan values, and strange fluctuations pattern.

stex2005, 2016-02-22:
Hi,

I'm using a DHT22 sensor with an ESP8266-201.
The ESP is a remote sensor, battery supplied and it goes deepsleep every 10 mins.
Sometimes the published data become nan until I reset the esp or they show strange fluctuations patterns as shown in figure.
I can't understand what it could be.
Maybe i'm not waiting enough time for the DHT data acquisition?  (i'm waiting 2 secs)
Maybe is a power supply problem? 4 AA batteries stabilized at 3.3V.
Maybe is a wire problem?
Maybe is a defective DHT module?

[image]

markwilliamson05, 2016-03-12:
I've been having issue with the DHT11 and ESP-01. When I have it hooked to the ESP I get inconsistent readings. When I use with Arduino, the reading are far more consistent. Any thoughts? Here are some snapshots of serial plotter to show difference with same sensor in same room.

With ESP:
[image]

With Arduino:
[image]

franc-carter, 2016-03-13:
I'be had similar issues with the DHT-22, my guess (without solid evidence)
is that it's PSU related. When I swapped to a NodeMCU with a nice well
design PSU on it then readings became much more stable

cheers

markwilliamson05, 2016-03-13:
Can you recommend a PSU? What did you end up using?

franc-carter, 2016-03-13:
I ended up not using the esp-11, I decided that it was easier (and possibly
cheaper) to swap to a NodeMCU that has an onboard PSU designed by
professionals.

For the esp-01 I was using a simple ebay 5v-3.3v regulat which cost $0.99
and when I looked at it with a scope there was pretty wild voltage
fluctuations

cheers

markwilliamson05, 2016-03-16:
Thanks so much for the advice to look at the power supply. I used a much more stable supply and now my readings are very consistent.

franc-carter, 2016-03-17:
No problem - glad it sorted it out.

It's also great to have someone else confirm the fix

cheers

markwilliamson05, 2016-03-17:
Now I'm not so sure. I've left it running and I'm now seeing spikes. I will keep testing.

franc-carter, 2016-03-17:
Let me know how you go - I'm quite interesting in why this happens.

On a semi-related note, that might be interesting. I have an esp-01 as a
'sensor node' that works quite well, except when I do my washing. The
esp-01 is on the same circuit as the washing machine and it looks like the
motor produces spike big enough that they get through all the PSU filters
and hang the esp

cheers

stex2005, 2016-03-17:
Pointed out that also in my case it was a PSU problem, batteries were to low to stabilize 3.3V.

franc-carter, 2016-03-17:
Ok, thanks more info pointing to PSU

My understanding is that the esp can draw a large spike of power when it
transmits - this might be dropping the voltage below the functional level
for the DHT (But - I am not an ElecEng)

cheers

markwilliamson05, 2016-03-17:
I will do more testing with different power supplies and post my findings.

markwilliamson05, 2016-03-20:
What's everyone using for pull up resistor on dht11 data pin?

markwilliamson05, 2016-04-01:
So after trying 3 different power sources, I'm still getting the same results (see below) with my DHT11. I'm now using arduino to read values and esp8266 to send message via MQTT. Any thoughts guys?

[image]

greggubben, 2016-04-02:
Mark,

I use a 6.8K resistor.  The DHT-11 datasheet suggests a 5K resistor.  

Are you using a capacitor across the ESP8266 power intake?  I saw some comments saying something about adding a capacitor to stabilize power.  I use an AMS1117 3.3v regulator to provide 3.3v to the ESP8266 and DHT-11.  The AMS1117 3.3v datasheet suggests a 22uF tantalum capacitor which is what I used.

markwilliamson05, 2016-04-02:
I'm using a 4.7k resistor for the dht11 data pin and using a 10uF electrolytic cap across the output for my lm1117 3.3v regulator. It calls for a tantalum but I don't have one. I'm wondering if the pattern of my readings is just normal for the dht11. It seems to be quite a steady pattern of up and down.

greggubben, 2016-04-02:
Based on the scale in the latest picture of 5 days, that fluctuation of 5% humidity throughout the day may be normal for your conditions.  Mine bounces around a bit, but I attribute it to the furnace cycling during the day because I have a humidifier added to my furnace.

markwilliamson05, 2016-04-02:
I tried correlating the pattern to the furnace running but it's definitely not that. I'm going to go back to plain old Arduino (taking esp8266 and lm1113 voltage regulator out of circuit) and monitor the readings via serial monitor to see if there is a major difference m.

markwilliamson05, 2016-04-26:
So I've finally determined that it is the ESP8266 itself that is producing the noise which is interfering with the DHT sensor. I've used countless power supplies and they are all fine. It is when the esp8266 is powered on the same power supply that the issue is occurring. The only way around this is to separate the power supplies or to determine exactly where the noise is coming from and try to filter it out. I don't have an oscilloscope so I can't determine how to filter. Any ideas?

KKKSzili, 2016-07-21:
I have the exact same problem, and figured out after getting nan values from the sensor, i unplug and plug back the power supply for the DHT22 sensor. I do not have any isssues with this lib on arduino.

[http://community.blynk.cc/t/nodemcu-esp8266-dht22-blynk-run-nan-issues/7074](url)

waynepiekarski, 2016-08-23:
I am having problems where after a day or so, I start getting NAN values from the sensor. If I power cycle the device things are usually fine for another day and then I get NAN again. The strange part is that the hardware worked for a few weeks continuously before it started failing. So I'm not sure if I'm running into hardware problems. This set up has two DHT22 sensors on an ESP8266 Huzzah board. Interestingly enough, I have another Huzzah with only a single DHT22 and that one has ran perfectly for months. Both of them are using USB phone chargers as power supplies for the 5V, and I've tried different chargers but it doesn't seem to help.

fantasymaker-cn, 2016-08-24:
**Just share my experience**
I had same `NAN` problems for a while before. At last I found it was some code disturbs DHT reading method. 
For example, I'm using `ESP8266WIFI.onEvent()` to monitor wifi status. Every time an event occurs, I print a log to Serial. This result to `NAN` while DHT reading. After I delete print log statement, everything goes smoothly.

Theteendevs, 2016-09-08:
I've got that issue too now....after a day it'll just randomly start giving NAN values......

Diaoul, 2016-10-18:
Same issue, here. I start having `nan` values after a day or so. I checked the heap, there is no memory leak, I removed any Serial interference without success and my power supply is fine.
Maybe this issue has to do with some update in Arduino/ESP8266/DHT-sensor-library?

pidiman, 2017-02-14:
Hello everyone. I make my DHT22 + ESP8266 - 12F temperature meter. This TMeter send every 15minutes temperature and humidity to my mysql database.

Everything works fine until I change power source from usb cable connecting to PC to 5V powerbank with voltage regulator to 3.3V. 

Temperature meter work but every value of temperature and humidity is **NAN**. Can you help me with this problem? 

Thanks a lot :)

DrGlas, 2017-04-20:
I also had fluctuating values from DHT11 on a Rpi 2 that was running from a 5V/2.1A power bank (iphone charger). Switched to a regular PSU 5V/1A and the DHT11 values immediately got very stable.

nikitz, 2017-05-27:
Check the datasheet 4.2 
Power supply 3,3 -5V . Recomended power supply 5v.
Iam having the same nan readings when powered by3,3v only @stex2005 
[https://www.google.bg/url?sa=t&source=web&rct=j&url=http://akizukidenshi.com/download/ds/aosong/AM2302.pdf&ved=0ahUKEwi1xOWFl4_UAhXH7xQKHVesAocQFggxMAM&usg=AFQjCNFQnhB8Mi7AhfGv5EU_nNxzoAgsjg&sig2=EHLGuMYG3N2kgYq1BhRINA](url)

jtippman, 2017-06-12:
I'm using a MKRZero which says it runs all pins at 3.3V. While the USB is 5V, the battery is 3.7V, and I am seeing the same nan output when reading temperature and humidity on the battery power. Maybe the entire board needs 5V and there isn't enough juice to power the sensor at this point.. I have basically everything configured as in the weather station project page on arduino. I found lixplor's comment on this page interesting. I wonder if my modified code is causing problems? I tried the original code on the project page, no luck. Also, I agree the possibility a dht update caused issues. I tried a few prior dht releases and this still did not fix it.

tl/dr: MKRZero with dht22 works great on usb power, nan on battery power. No known fix that I can find.

mrburns42, 2017-08-23:
I am having the same issues running from an ESP8266.   For testing purposes, the code Sleeps for 1 minute, then wakes up connects via WiFi and then measures the temperature and humidity before transmitting via MQTT.    
It is running from a LiIon cell with a +3.3V LDO.    It runs for over 12 hours and then it can no longer read the DHT22.   When I start getting the NAN, if I wait until the ESP8266 is asleep and then unplug and replug the DHT22, then it starts working immediately on wake up.   Note:  the readings are stable when it works.  Battery charge level is good and it works for many, many, many additional hours after replugging the DHT22.  I suspect running the DHT22 at 5V is only masking the true problem.
I turned on Debug Printing in the DHT.cpp.   There seems to be a potential miscommunication in DHT::expectPulse().   The variable count is initialized to zero.  However, zero is also the value that signifies an overflow.  If this method is called and the signal is already in the opposite state, then it will return a timeout when none occurred.   I changed the init value of count to 1 instead of zero.  Then I added a Warning printout at the higher level if expectPulse ever returns a value of 1.    Below is the debug messages at the point of failure.   The last successful reading and then the first bad reading.    From these results, it appears that the start signal code is a little too trusting.  It assumes that the value is still LOW before entering the first expectPulse.  If the DHT22 missed or did not respond to the wakeup, then code execution on the ESP8266 and the DHT22 will be out of sync.    If when expectPulse() returns a '1', if the DHT22 could be issued a reset command then I believe the problem could be overcome.  
*****************************************************************************
DHT22 test!
Connecting to Pi_AP
......
WiFi connected IP address: 192.168.42.11
Max clock cycles: 80000
Received:
1, 82, 1, 7, 8B =? 8B
Connecting to 192.168.42.1 as esp8266-18:fe:34:9b:d8:e-a2
Sending payload: Humidity,38.60,Temperature,79.34 
Publish ok
8þÿ 
DHT22 test!
Connecting to Pi_AP
......
WiFi connected IP address: 192.168.42.11
Max clock cycles: 80000
 Warning! Signal already High when checking start signal low pulse.
Timeout waiting for start signal high pulse.
Failed to read h from DHT sensor!
Failed to read t from DHT sensor!
Failed to read f from DHT sensor!
Failed to read from DHT sensor!

mrburns42, 2017-08-29:
For the ESP8266, I believe I know what is happening.   When the ESP8266 goes into Deep Sleep,  it completely reboots upon wake-up.   Deep Sleep is the lowest power state and when using that, there is no way to avoid a complete reboot.   This ESP8266 reboot includes it running its Bootloader code.   This Bootloader code sends some data out both GPIO_0 and GPIO_2.    I had the DHT22 connected to GPIO_2 as the only available IO on the ESP-01.    The screen shot below shows the activity on GPIO_2 when code restart after a Deep Sleep wakeup.   When I took this scope shot, all DHT22 code and all WIFI code was commented out.   The ESP-01 was only pausing for several seconds, then going into Deep Sleep.

[image]

My theory is these pulses on Startup are putting the DHT22 into some undocumented mode.  Perhaps it thinks new firmware is being loaded.   Or perhaps it is a mode where calibration data is loaded.  Regardless, once in that mode, the DHT22 will not respond to normal commands and the system is hosed until I power cycle the DHT22.  
Both GPIO_0 and GPIO_2 show activity during the bootloader.   Therefore, neither of these pins are suitable for use with the DHT22 when running the Deep Sleep.   As a test, I switched the DHT22 to use the RX pin of the ESP-01.     From within the Arduino code, that pin is considered GPIO_3.  
#define DHTPIN 3
I have run the system for 24 hours now without any NAN values.   Since it wakes up and measures the temperature every second, that is over 1400 successful accesses.   Previously, it would fail within 8 to 12 hours.  
Note GPIO_3, aka RX, is needed to load code onto the ESP-01.  Therefore, I connected it to a three pin header.  I can jumper two pins to program and then power down and jumper the other two pins to access the DHT22.

heckler, 2017-11-15:
For whatever it is worth, I've observed the following:

## Setup

I have both a DHT-22 and a DSB1820 on the same board, as I assumed initially that the Dallas sensor would be much more precise than the the DHT (turns out they are always within .5 ºC of each other, sweet!)

This is the setup where I'm testing it:

- NodeMCU v1.0, powered from an USB port on my PC
- DHT-22 on node pin D5 with 10K pull-up
- DS18B20 on pin D7 with 4K7 pull-up
- Arduino Core 2.4.0-rc1
- DHT Sensor lib v1.3.0
- OneWire lib v2.3.3
- DallasTemperature lib v3.7.6

## Observed behavior

- If I run the DHT libirarie's DHTTester example, I can let it run for as long as I want an no single NaN is reported (the sample would print "Failed to read from DHT sensor!" in case any NaN was returned from the lib).  This lets me conclude that the sensor and board power are OK.
- If I run my own sketch, which is sleeping for 10s, then reading the DS18B20, next reading the DHT-22, and finally printing the values out, I get the result that every 5-10 readings I get a NaN from the DHT-22 lib.
- Considering that the sample sketch "just works", I've decided to comment-out all code related to the DS18B20, and "bingo!", no more NaNs.  Something is at work here apparently.
- I tried disabling interrupts (nointerrupts()/interrupts()), but that didn't change anything (and in any event, the DHT library already lock interrupts during the time-critical portion of the data read)
- Next I experimented with the order, tried reading the DHT sensor first, and then reading the DS18B20 after that.  The result from this was also **no NaN readings** after a long time testing

## Conclusion

Not yet 100% sure what is the deal here, but definitively my problem is software-based.   For anyone experiencing the occasional NaN reading, I can only suggest that you try the library example "as is" and see how that behaves on your board+sensor.

Craig1516, 2018-01-15:
In addition to the power supply solutions mentioned above I found that adding  dht.begin(); to the setup routine helped.  I was getting about 50% NAN readings without this statement so some success can be had without the statement.  

My power supply problems were coming from my USB/TTL converter despite no WiFi transmission.  The ESP is a 3.3V device and that is at the bottom of the DHT voltage range (3.3V to 5.5V).  So perhaps the best approach is to separately power both devices.

Yves1812, 2018-01-27:
A big thank you to MrBurn42.
I had the same issue with a Wemos D1 mini pro. I started using GPIO2 had random nan, tried the next pin - unfortunately GPIO0 with no improvement - until I red your analysis.
Now running like a charm on GPIO12.

Diaoul, 2018-01-27:
I'm now using my own library due to multiple issues with others. I've
achieved 0% error rate with it after 30k+ readings.

I had another major issue with the libraries being way too slow and this
caused many issues with my LED animations.

My DHT is powered with 3.3V and 4.7k pull-up resistor with an ESP-12E.

The library is called DHTNew and available on PlatformIO. Feel free to try
it out and see if there is any improvement for you.

waspinator, 2018-02-15:
@Diaoul is yours related to https://github.com/RobTillaart/Arduino/tree/master/libraries/DHTNEW ?

DebtFromAbove, 2018-03-22:
Hi guys,

I've been at this problem as well for a while now and I thought I'd share my experience in the hope that someone else finds it useful.

HW setup: ESP-12E bare-bones module + DHT22
SW: Arduino IDE with ESP8266 board library 2.4.0-rc1
Adafruit DHT library
Power: nominal 3.7V 18650 Li-on cell -> HT7533-3 3.3V 100mA LDO + 1000uF + 0.1uF caps
DHT22 data line attached via 10K pull-up. Data line connected to GPIO4

Problem #1:
First I tried doing readings from the DHT22, sending them via UDP to a server then doing deep sleep for 30 seconds, wake-up and repeat. Like many have reported I had problems where it would not read the sensor on reboot. I resolved this by connecting the DHT22 VCC pin to a GPIO on the ESP, not directly to the power source.* I set the pin up as an output, normally pulled low. After the ESP boots the pin is pulled high, delay(1000)**  read sensor, then pull pin low before going to sleep again. Now it will return a reading successfully after reboot

Problem #2:
DHT22 readings randomly failed. at fairly sporadic spacings, read attempts would randomly fail to return a result. The dreaded 'NaN' response. Reading around about this issue I found the culprit to be in the Adafruit DHT code. Once I commented out a few of the lines as per this issue (https://github.com/adafruit/DHT-sensor-library/issues/48) my readings became MUCH more reliable.

Problem #3:
DHT22 readings started to fail more often as the battery voltage dropped (but still well above the dropout range of the LDO e.g. 3.6V). Deciding to actually read the datasheet I found some information that said the VCC voltage should be kept stable for at least 2 seconds before attempting to take a reading. Going back to my solution for problem #1, I was using a delay(1000) after setting the GPIO pin high that was connected to DHT VCC. I wasn't leaving it long enough before trying to read. Most of the time this seemed to be fine, but something about the battery voltage starting to tail off meant that this impatience caused it to fail to read more frequently. Increasing this to delay(2000) fixed this issue.  Below are my humidity read results over 48 hours before the fix, where the line drops to 0 is where it returned a NaN response:
[image]

Problem #4:
Fluctuating temperature (and humidity) readings. After fixing the above problems I thought I was doing pretty well. I added battery voltage monitoring via the analog A0 pin, via a 320 / 100kOhm voltage divider and watched the readings come in. After a while I started to notice that there was a nice normal curve to my daily results from morning to night. However fairly often I would get spiky readings, sometimes over 2C higher (nearly always higher for some reason), then the following reading would drop back down to being in line with the others. See graph below
[image]

Knowing the temp in my house didn't fluctuate that much I knew there was another issue at play. My first thought about this problem was that it was probably a power issue. A 100mA LDO does not provide a lot of juice for the power hungry ESP. But I had a large 470uF cap plus a 0.1uF ceramic cap across the VCC and GND lines of the LDO, so I figure it was probably enough to keep the juice flowing for those pesky boot-time and WiFi RF power spikes. I added a couple of caps across the DHT22 VCC+GND pins thinking that it was probably the sensor drawing too much current during a read which was brining the battery / LDO output voltage too low for a reliable read. After another day of testing this showed no improvement. I knew I wanted to figure out if it was the ESP causing the issue, so I set about trying to disable the WiFi modem during sensor reads, just to ensure there weren't any disruptions to the power. I found this blog which detailed how to temporarily disable and then enable the WiFi modem (this is different to the WAKE_RF_DISABLED command you can provide when instructing it to deepSleep. It seems the ESP WiFi can't be re-enabled again after this without another call to deep-sleep)
https://www.bakke.online/index.php/2017/05/21/reducing-wifi-power-consumption-on-esp8266-part-2/

Now my ESP wakes up, disables the WiFi modem, enables the DHT, waits 2 seonds, takes a sensor reading, disables the DHT power pin again, enables and connects to wifi, transmits the result and goes back to sleep. My result now is a very stable, fairly accurate temperature/humidity measurement as you can see from this last graph of 24 hours:
[image]

This is my experience with DHT22 and ESP-12E modules. I still have work to do, next is a cycle count using RTC memory to check if it really woke from deep-sleep or if power failed, MQTT publisher and then OpenHAB integration

I hope you find the experiences and fault resolution steps useful for your own projects. Let me know if there is anything else you would like to know about my setup. I am not an expert by any means however.

* I want to figure out a way to put DHT22 VCC back on the battery, as I think it will ultimately be more stable directly on the battery as this will always be at least ~150mV higher than the LDO output, so should start failing to provide sensor reading later if connected directly to the battery. In this case the ESP will start becoming unstable before the DHT22

** More on this later, it turns out delay(2000) was actually needed instead

phuzybuny, 2018-03-25:
@mrburns42 My problem with the Wemos D1 mini and DHT22 was exactly as you described. Reconfigured my data pin and it solved it. Thank you!

GiFe71, 2019-01-04:
@DebtFromAbove, very interesting. You are completely right. I'd like only to add that your solution seems to not work with ESP-01 due to the fact that you cannot pull down the pins GPIO0 or GPIO2, otherwise you'll boot the module in the wrong mode i.e. flash. The DHT22 has a very low power consumption, so it can be always power on. I adopt the solution using ESP-01s, DHT22 always on, reads the sensor using RX(configured as GPIO3). It has a power consumption of around 75mA when reads the sensor and transmits data and 0.035 mA during Deep sleep. Anyway I'll implement your suggestion about wi-fi sleep during sensors reading.
Thanks

GiFe71, 2019-01-05:
I was wrong, it works using GPIO0 as the sink on the ground of DHT22, and GPIO2 for data, pull up required.
Power consumption in sleep 0.012 mA, reading 40 mA transmitting 75 mA.

PaintYourDragon, 2019-02-14:
Try newest library - reopen if still an issue.

closed 2019-02-14 by PaintYourDragon

djmaze, 2020-01-31:
On my ESP-12E, I had to switch my DHT22 to pin GPIO14 (D5). Neither GPIO2 (D4) nor GPIO3 (RX) would work after deep sleep, but GPIO14 does.   

Thanks for the hint @mrburns42.