adafruit/DHT-sensor-library #62: DHT Library 1.3.0 Does NOt Work

sghazagh, 2016-10-28:
Hi,
I can compile my project properly with DHT library 1.2.3.
However, after updating my library to version 1.3.0, this error raised and I cannot compile my project using DHT22 sensor:

//////////////////////////////////
C:\Users\xxxxxx\Documents\Arduino\libraries\DHT_sensor_library/DHT_U.h:25:29: fatal error: Adafruit_Sensor.h: No such file or directory

 #include <Adafruit_Sensor.h>

                             ^

compilation terminated.

exit status 1
Error compiling for board WeMos D1 R2 & mini.
/////////////////////////////

sghazagh, 2016-10-28:
I could manage to fix the compile issue by adding this :  https://github.com/adafruit/Adafruit_Sensor

But why we need it in first place and why cannot add it from Library Manager ???
(I manually added this to work)!!!!

stelgenhof, 2016-11-01:
I am experiencing the same and adding `#include <Adafruit_Sensor.h>` does resolve the issue. It seems a requirement for the DHT Unified sensor, however I am using still the previous way (using DHT.h).

oy5tein, 2016-11-03:
1.3.0 is broken here to (and adafruit sensors is not in library manager)
In DHT sensor lib in the lib manager I clicked on select version 1.2.3 and my old sketches started working again.

sanjaybhagia, 2016-11-05:
I even tried version 1.2.3 (also with Adafruit_Sensor.h) but nothing seems to work for me. Its simply fails to read the data from the sensor.

emilfihlman, 2016-11-12:
Broken here, too. Not cool pushing broken things to the public Arduino libraries...

ViliusKraujutis, 2016-11-28:
I fixed this issue by saving [this Adafruit_Sensor.h file](https://github.com/adafruit/Adafruit_Sensor/blob/master/Adafruit_Sensor.h) to new folder `$ARDUINO_HOME$/libraries/Adafruit_Sensor/Adafruit_Sensor.h`.

yusufbulentavci, 2016-12-24:
Including in this order solves the problem
#include <Adafruit_Sensor.h>
#include "DHT.h"_

microbuilder, 2016-12-26:
You should be able to install the missing Adafruit_Sensor library from the Library Manager once you add the Adafruit .json file: 
[image]

closed 2016-12-26 by microbuilder

salmanfarisvp, 2017-01-06:
thanks @ViliusKraujutis. problem solved 
@microbuilder can you please add the .json link here...

vijiarun, 2017-04-10:
please can anyone help me solving this??? I  too have the same error after downloading the library.

vijiarun, 2017-04-10:
I am getting this error please help me to solve this error: 'dht' does not name a type
dht DHT;

marcosavini, 2017-04-17:
Downgrading the lib to 1.2.3 worked for me.

pucbaldwin, 2017-04-17:
Download the "Adafruit Unified Sensor" library from the library manager - this worked for me.

JeremyOne, 2017-05-03:
I also had a problem with this, downgrading to 1.2.4 worked after I found this thread. It is a little distressing to do the automatic library upgrade in the Arduino IDE and suddenly have a broken compile.

akeilox, 2017-06-21:
i had this exact same problem, and switching the DHT lib to 1.2.3 does solve the problem of compile error and compiles fine.

Didnt try the route of adding Adafruit Unified Sensor as suggested by others.

rkveale, 2017-08-09:
As was previously mentioned, download the "Adafruit Unified Sensor" library from the library manager.
http://www.arduinolibraries.info/libraries/adafruit-unified-sensor

chipytux, 2017-10-27:
Downgrade to 1.2.3 Version to solve your issue.

jimgitonga, 2018-01-03:
Download the "Adafruit Unified Sensor" it worked out for me https://github.com/adafruit/Adafruit_Sensor

chipytux, 2018-01-09:
Thank you, this work.
> Download the "Adafruit Unified Sensor" it worked out for me https://github.com/adafruit/Adafruit_Sensor

hoegge, 2018-01-10:
But that library is not part of the Arduino IDE library manager, right? Which means a lot of people won't find it. Shouldn't it be included there as a part of the library using it?

geofbaum, 2018-01-10:
@hoegge  I don't know which version of the IDE you might be using but I can see the Adafruit Unified Sensor in the library manager. It was the second to last one on the list that I got while filtering the search with 'Unified Sensor.'

hoegge, 2018-01-10:
@geofbaum 1.8.5 - sorry tried again, now it is there. Must have made a typo or something - sorry for the inconvenience

geofbaum, 2018-01-10:
@hoegge  No inconvenience at all! Glad you were able to find it!

Gacon17, 2018-04-08:
Hi sghazagh,

I experienced the same problem, to remove that issue you should re-update the older version such as 1.2.3
It will work as your desire.

salmanfarisvp, 2018-04-08:
For using [Adafruit DHT Humidity & Temperature](https://github.com/adafruit/DHT-sensor-library) library You must have the[ Adafruit Unified Sensor Library  ](https://github.com/adafruit/Adafruit_Sensor) in Arduino libraries.

AgusTriwahyudin, 2018-07-06:
yes. you can downgrade to version 1.2.3

Hammerschmidt, 2018-08-06:
You help me too. Thanks

otmezger, 2019-01-20:
It sems that 1.3.1 still has the same problem. Downgrading to 1.2.3 works perfectly

sachendra003, 2019-04-25:
Download the "Adafruit Unified Sensor" library from the library manager  - this worked for me.
[image]