adafruit/DHT-sensor-library #65: Cannot measure below 1 deg celcius. Sensor or code?

nitech, 2016-11-09:
I run the standard code, included in the [example folder](https://github.com/adafruit/DHT-sensor-library/tree/master/examples/DHT_Unified_Sensor). I put the sensor outside, to measure the current freezing norwegian temperatures, but look what it outputted: 

Temperature: 1.00 *C
Humidity: 58.00%

I notice two tings: 

1. There is about -6 deg c outside right now. Not +1
2. Where are the decimals?

magtimmermans, 2016-11-30:
I have the same thing.

nitech, 2016-11-30:
It turns out that my problem was the DH11 sensor which simply doesn't measure below 1 degree Celsius. Probably obvious to the manufacturer, not to be. Meant for indoor measurement? 

Also the dh11-sensor does not measure decimals.

magtimmermans, 2016-11-30:
Aha, but u have dth22 and will also jot show negative values.

nitech, 2016-11-30:
Right - then you have a different problem than I. The DHT22 is supposed to measure down to -40 degrees celcius.

geofbaum, 2016-11-30:
Is this issue solved or not? I haven't had a problem with any of the DHT22's that I have that have been used in freezers measuring below 0 degrees. As you stated and as is written in the products datasheet the DHT11s will not report below 0 degrees and only has a resolution of whole degrees.

magtimmermans, 2016-11-30:
Not for me and I have dht22.

geofbaum, 2016-12-01:
So you are you utilizing the Unified version of the code or the older original version? Also are you using the 10K resistor between the VCC line and data line, or a board with the DHT sensor and resistor already on it? I'm just trying to see where an issue could be happening as I don't know what your setup is, or which version of the DHT22 you're using. I just ran another test with a version of the DHT22 on a WeMos shield which has a resistor built in. It was sitting in a freezer if you'd like to check. the values I got from it.

https://thingspeak.com/channels/184956

The code it is running from is based off of the older code before the Unified version, but I can try to reload it and run it with the Unified version of the code.

magtimmermans, 2016-12-01:
Hi, I am very new to all this. I have indeed a shield including the resistor. I am not sure what you mean with the unified version. I downloaded using the library manager from the Arduino software the 1.3.0 version. For me testing is a little bit difficult as the device is mounted outside and I am depending on the weather.

geofbaum, 2016-12-01:
No worries, and I apologize if I came off as being rude in anyway. The link that nitech posted original points to the file in the Example folder of this repository for the Unified sensor code. The file is specifically called DHT_Unified_Sensor.ino so I did not know if you were using that example code to test your device. There is also the DHTtester.ino that is in the Example folder which uses the older code, it still works and it's the one that I've used more so as I haven't migrated all of my code over to the Unified Sensor code that Adafruit is pushing to make using different sensor types all work with a single type of code.

Either way I'll attach a simple example of the code that I'm using if you want to compare it to what you are using to see if you can spot what might be causing your issue of not seeing negative values.

#include <DHT.h>
#define DHTPIN D4     // what pin we're connected to
#define DHTTYPE DHT22   // DHT 22  (AM2302)
 
DHT dht(DHTPIN, DHTTYPE);

void setup() 
{
    Serial.begin(9600);
  delay(10);
  dht.begin();
}

void loop() 
{
  // Wait a few seconds between measurements.
  delay(60000); // 2 minute delay = 120000 -- 1 minute delay = 60000 and so on

  float h = dht.readHumidity();
  float t = dht.readTemperature();

    // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t) || isnan(f)) 
  {
    Serial.println("Failed to read from DHT sensor!");
    return;
  }

  Serial.print("DHT22 Humidity: ");
  Serial.print(h);
  Serial.print(" %\t");
  Serial.print("DHT22 Temperature: ");
  Serial.print(t);
  Serial.print(" *C ");
  
}

magtimmermans, 2016-12-01:
Thank you very much!, This code is working need to check what is wrong with my code. One thing I already see is that my wait time is very short. But my positive values where ok. Anyhow the issue can be closed as this is my code and not the library! Thanks for pointing me in the correct direction.

geofbaum, 2016-12-01:
No problem, I'm glad I could help.

closed 2016-12-24 by ladyada