adafruit/DHT-sensor-library #69: Release 1.3.0 don't work for ESP8266

DaniloFeng, 2016-12-28:
I'm trying to compile for ESP8266 (ESP12E) your library, but for release 1.3.0 don't work.
Appears this message:

`Using library DHT_sensor_library at version 1.3.0 in folder: C:\Users\...\Arduino\libraries\DHT_sensor_library 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
`

But, when I use release 1.2.3 is work fine!

northalpha, 2017-02-06:
Same happening here on a Wemos D1 mini

Ostapas, 2017-03-01:
you have to install additionally adafruit unified sensor library

rafaellehmkuhl, 2017-12-14:
Can confirm it only works with 1.2.3 or below. Wemos Mini D1 here.

Also, reload Arduino software after changing library from 1.3.0 to 1.2.3 or it won't reload the library,

PaintYourDragon, 2019-01-15:
Examples & README have been updated to clarify the Unified Sensor Library requirement.

closed 2019-01-15 by PaintYourDragon