adafruit/DHT-sensor-library #71: ESP8266 NodeMCU (ESP12E Module) don't work

Jcfurtado, 2017-01-16:
I'm trying to compile for ESP8266 NodeMCU (ESP12E) your library, but don't work.
Appears this message:

Arduino: 1.8.1 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

/opt/arduino-1.8.1/examples/DHT_Unified_Sensor/DHT_Unified_Sensor.ino:10:29: fatal error: Adafruit_Sensor.h: No such file or directory
 #include <Adafruit_Sensor.h>
                             ^
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

khjoen, 2017-01-17:
Hi JcFurtado,
did you install this library https://github.com/adafruit/Adafruit_Sensor?
According to README.md, it is now required for this library.

Pascal

raxpa, 2017-01-27:
I've got the same problem here... doesn't happen with version 1.2.3 with works perfectly

khjoen, 2017-01-27:
Hi Raxpa,
You need the Adafruit_Sensor library even though there is no reference to it in the old DHT code.  There are the new DHT_U files that reference it and the IDE compiles all the files in the directory, as a consequence we need to install the Adafruit_Sensor lib.  To fix, download the library and extract it in the  libraries directory under your sketchbook directory.  You should be good to go after doing so.

Pascal

raxpa, 2017-01-31:
thank's I'm actually using the old version without any problem

Jcfurtado, 2017-01-31:
Thank´s everyone for help

mohamedbenalima, 2018-10-12:
hello guys i have the same problem and i installed addfruit librairy but it dosn't work 
is there any other solution ? 
"compilation terminated.

exit status 1
Erreur de compilation pour la carte NodeMCU 1.0 (ESP-12E Module)"

closed 2019-02-14 by PaintYourDragon