adafruit/DHT-sensor-library #72: Include <Adafruit_Sensor.h> error

ViktorKovalev3, 2017-02-12:
Add [Adafruit_Sensor.h](https://github.com/adafruit/Adafruit_Sensor) in to the DHT sensor library.

torabian, 2017-06-29:
I am having this issue. I've addded DHT but my build fails, since this file is not found. How can I do that?

torabian, 2017-06-29:
I am trying to copy the file manually intodht sensor folder in my travis build hope to get results from there.

idelara, 2017-07-16:
@torabian You have to manually copy and paste the file called "Adafruit_Sensor.h" from the repo @ViktorKovalev3 mentioned. Download (or clone) the repo, go to your Arduino library folder, locate the DHT library folder, and copy such file there.

jimsy3, 2017-07-23:
i just installed somekind of Unified Sensors library(by adafruit) and then it worked for me

rafagildin, 2017-10-12:
i need a help please, when i put to work this code this appear in IDH arduino:"Adafruit_Sensor.h: no such file or directory " what i can make ?

Jotschi, 2017-11-19:
@rafagildin You need to download and install the https://github.com/adafruit/Adafruit_Sensor library as well.

closed 2019-01-12 by PaintYourDragon

LRagji, 2019-04-20:
This is still a pain, not sure why was this closed Library management in arduino has not reached full potential of other package managers like npm where dependencies of dependencies are installed automatically for a new comer its still going to be a problem as the library is not available in library manager in arduino

ladyada, 2019-04-20:
you can comment here to support Arduino's efforts
https://github.com/arduino/Arduino/pull/8600

lekimtek, 2021-11-04:
Same problem in PlatfomIO, I resolved by copy this in sketch.ino;
#include <Adafruit_Sensor.h>
this resolve problem.
I dont now happens this, because  "#include <Adafruit_Sensor.h>" is in the library but not link wen compiling.