adafruit/DHT-sensor-library #73: RedBear_Duo with DHT AM2302

jadeye, 2017-02-19:
I have a `RedBear_Duo` and I am trying to use the `DHT AM2302` with this library.
I keep on getting this error, **any idea how to resolve this**..
```
In file included from /home/user/Arduino/dht_try/dht_try.ino:12:0:
/home/user/Arduino/libraries/DHT_Grove_Temperature_And_Humidity_Sensor-master/DHT.h:16:2: error: #error "CPU SPEED NOT SUPPORTED"
 #error "CPU SPEED NOT SUPPORTED"
  ^
Using library DHT_Grove_Temperature_And_Humidity_Sensor-master in folder: /home/user/Arduino/libraries/DHT_Grove_Temperature_And_Humidity_Sensor-master (legacy)
exit status 1
Error compiling for board RedBear Duo (Native USB Port).
```
**Thank you**

geofbaum, 2017-02-19:
You will have to adjust the values that are accepted for CPU speed in the version of the DHT.h file you are using. I think that board from what I found online is 80Mhz so my advice would be to either write a new line similar to the ones already in the file for the 8 Mhz and 16 Mhz CPUs or change the upper limit on the 16 MHz section to a value that would allow for 80Mhz. I believe the lines you need to look at are 9 to 17 in that version of the file.

jadeye, 2017-02-20:
@geofbaum , thanx for your help...
According to the specs page its CPU runa at`120Mhz` - [redBear_Duo_Spces](https://github.com/redbear/Duo/blob/master/docs/duo_introduction.md#features)
But doing a little investigation into the code:
```
// 8 MHz(ish) AVR ---------------------------------------------------------
#if (F_CPU >= 7400000UL) && (F_CPU <= 9500000UL)
#define COUNT 3
// 16 MHz(ish) AVR --------------------------------------------------------
#elif (F_CPU >= 15400000UL) && (F_CPU <= 19000000L)
#define COUNT 6
// 120 MHz(ish) AVR --------------------------------------------------------
#elif (F_CPU >= ____________UL) && (F_CPU <= ____________L)
#define COUNT __??
#else
#error "CPU SPEED NOT SUPPORTED"
#endif
```
I would appreciate it if you could point me in the right direction:
1. what are or better yet, how do you calculate the range for `F_CPU >= __UL && F_CPU <= __L` OR `UL` (at the end of the line???
& what about the `COUNT` value??
[understanding F_CPU](http://www.avrfreaks.net/forum/understanding-fcpu)
[ARM Cortex-M3 MCU w/ 1 MB Flash, 120 MHz CPU - Clock -  4 to 26 MHz crystal oscillator](http://www.st.com/content/ccc/resource/technical/document/datasheet/bc/21/42/43/b0/f3/4d/d3/CD00237391.pdf/files/CD00237391.pdf/jcr:content/translations/en.CD00237391.pdf)
Is it a simple calculation of 120 * something??
2. I might be mistaken but does the crystal speed have anything to do with this??
[](http://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f2-series/stm32f2x5/stm32f205rg.html)

Thanx

geofbaum, 2017-02-21:
The F_CPU range values are simply related to the speed it's trying to accommodate as you can see from the code for 8MHz it's trying to see values between 7.4-9.5, for the 16MHz it's similar in that it's looking for 15.4 to 19.0, as those UL values are simply values in Hz instead of MHz. So in that sense if it's based off of the 120 it's simply changing it from 120 Mhz to 120000000 Hz. For me I'd simply check if 121000000 worked as the reason I brought this up was from seeing other people with boards or makeshift boards using something not within the "normal" arduino spectrum. Should it be related to the crystal oscillator, probably yes according to one of your links but every thing that I've seen related to an STM32 relates to that 120 value. Actually even here in this almost identical board the F_CPU value is set at 120000000 (https://github.com/adafruit/Adafruit_WICED_Arduino/blob/master/boards.txt).

As for the count value I wouldn't worry about it if you're adjusting an old value, if you're adding a new value I'd choose something that is equivalent to what is already there (3 for 8, 6 for 16, so then 9 for 24, 12 for 32, 15 for 40, etc. until you get I believe 45 for 120). Once again this is based off of an example I saw for someone running something off of 20 Mhz as opposed to 16Mhz so they just moved the second value in the range to account for it and left the Count Value the same. Personally I'd just check to see if it runs based on the same count value for the 16Mhz and then add the additional lines if necessary.

jadeye, 2017-02-21:
@geofbaum thank you again...
So I did get the code for the `CPU` to pass both `verification` and `upload` to `RedBear_Duo` but still I can't read anything off of the `AM2302` sensor.
Here it says [`3.Connection - Plug grove DHT pro to the socket D4 of grovepi.`](http://wiki.seeed.cc/Grove-Temperature_and_Humidity_Sensor_Pro/)
```
import grovepi

# Connect the Grove Temperature & Humidity Sensor Pro to digital port D4
# SIG,NC,VCC,GND
sensor = 4
```
Now if I am not mistaken `grovepi` mentioned there is of equivalent functionality as [`RBLink`](https://redbear.cc/rblink.html)

> The company also made a small baseboard called RBLink with  ST-LINK/V2 for programming the board, and 8 connectors compatible with Seeed Studio grove modules.
http://www.cnx-software.com/2015/11/18/redbear-duo-is-a-breadboard-friendly-wi-fi-ble-iot-board-based-on-ampak-ap6212-module-crowdfunding/

On the same page it says 

> With Arduino
> Connect the Temperature and Humidity Sensor Pro to D2 of Grove - Base Shield. Then plug Grove - Base Shield into Arduino and connect Arduino to PC using a USB cable.
> 

Anyways, on all cases, if I define:
```
#define DHTPIN 4
#define DHTPIN D4
#define DHTPIN 2
#define DHTPIN D2
```
I get no reading.

Do you have any idea why?? or how to go about fixing it??
Thank you

geofbaum, 2017-02-21:
No worries though I'll start this off by saying I don't use the Seeed boards or sensors myself so I don't know if my help will be of much use. Next, the connection of what pin you connect to is arbitrary so while those links might say connect to this Pin or that Pin it just matters that you make sure the correct Pin is defined in your code. To continue with that I just want to make sure you didn't do this and expect a returned value having the following won't help since all of them are defined as the same thing.

```
#define DHTPIN 4
#define DHTPIN D4
#define DHTPIN 2
#define DHTPIN D2
```

I don't think you did this as in your code you did correctly only define one DHTPIN value. If for some reason you're running more then one sensor you'd need to define the two or more separately like this:

```
#define DHTPIN  D1
#define DHTPIN1 D2
#define DHTPIN2 D3
```

My next question, which really shouldn't be an issue but could be is, are you actually connecting the Grove sensor directly to the RedBear Duo or are you connecting it through a shield like the RBLink you mention in your response?  Lastly, is it necessary for you to have the code calculating the temperature and humidity in the first half of your loop? 

```
    int a = analogRead(pinTempSensor);

    float R = 1023.0/((float)a)-1.0;
    R = 100000.0*R;

    float farenTemp=1.0/(log(R/100000.0)/B+1/298.15)-273.15;//convert to temperature via datasheet ;
    float celTemp = (farenTemp - 32) * 5/9;
    delay(1000);
    Serial.print("Celsius Temperature is: ");
    Serial.println(celTemp);
    Serial.print("Fahrenheit Temperature is: ");
    Serial.println(farenTemp);

    delay(100);
```

As well as the corresponding information in the area above the setup()?

```
// temparature v1.0
const int B=4275;                 // B value of the thermistor
const int R0 = 100000;            // R0 = 100k
const int pinTempSensor = A0;     // Grove - Temperature Sensor connect to A0

#if defined(ARDUINO) 
SYSTEM_MODE(SEMI_AUTOMATIC); 
#endif
```
I ask this because if you're using a plug and play sensor board like the Grove sensor the part of the problem that is really the point here is the DHT part of the loop and you might want to consider having that be the only part of your code until you get it setup. That's just my two cents though.

The rest of your code looks fine as it's just pretty much exactly the same as the DHTTest.ino files in this repository as well as the Seeed repository for the Grove Temp and Humidity sensor. I have used these sensors with both a Wemos board and a a Whisper Node with just the basic code without a problem so I'm just trying to see if there's something we're both missing.

jadeye, 2017-02-21:
@geofbaum - Thank you so much for your time and knowledge.

- I am only connecting a single `Grove DHT sensor`. The `DHTPIN` definition & examples I gave are only to point out I tried all 4 configurations.

- The code calculating the temperature (Fahrenheit & Cellcius) in the first half of my loop is for a different [Grove Temperature Sensor v1.0](https://www.seeedstudio.com/Grove-Temperature-Sensor-p-774.html) which actually still worx.
I am trying them both connected together through the `RBLink` just to make sure I'm still getting something...
 - Same goes for the part of code you display in the `setup()` part.
I am connecting the `Grove DHT Pro` through the `RBLink`.
 - Will give it a try with the code only for `Grove DHT sensor` (although I think this is not the issue, as the other `Temperature Sensor` is connected through an `analog` port)

What do you think??
Thank you.

geofbaum, 2017-02-21:
@jadeye  No problem, and I'd normally agree that the analog sensor shouldn't be causing an issue but it's always prudent to check if everything works separately before trying to combine things. I'd also suggest adding an additional line when you check to see if it was able to read the DHT sensor.

```
  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t)) {
    Serial.println("Failed to read from DHT sensor!");
    return;
  }
```

That `return;` was really the only thing that I noticed that was different between the basic code that I've used and your code for the DHT sensor.

jadeye, 2017-02-22:
@geofbaum I am not getting anything....

My `DHT.h` is:
```
#ifndef DHT_H
#define DHT_H
#if ARDUINO >= 100
 #include "Arduino.h"
#else
 #include "WProgram.h"
#endif

// 8 MHz(ish) AVR ---------------------------------------------------------
#if (F_CPU >= 7400000UL) && (F_CPU <= 9500000UL)
#define COUNT 3
// 16 MHz(ish) AVR --------------------------------------------------------
#elif (F_CPU >= 15400000UL) && (F_CPU <= 19000000L)
#define COUNT 6
// 120 MHz(ish) AVR RedBear -----------------------------------------------
#elif (F_CPU >= 120000000L) && (F_CPU <= 130000000L )
#define COUNT 45
#else
#error "CPU SPEED NOT SUPPORTED"
#endif

/* DHT library

MIT license
written by Adafruit Industries
*/

// how many timing transitions we need to keep track of. 2 * number bits + extra
#define MAXTIMINGS 85

#define DHT11 11
#define DHT22 22
#define DHT21 21
#define AM2301 21

class DHT {
 private:
  uint8_t data[6];
  uint8_t _pin, _type, _count;
  boolean read(void);
  unsigned long _lastreadtime;
  boolean firstreading;

 public:
  DHT(uint8_t pin, uint8_t type, uint8_t count=COUNT);
  void begin(void);
  float readTemperature(bool S=false);
  float convertCtoF(float);
  float readHumidity(void);

};
#endif
```
This is a modified version after many trials...
```
#include "DHT.h"

#define DHTPIN D2    // what pin we're connected to

#define DHTTYPE DHT22   // DHT 22  (AM2302)

// Connect pin 1 (on the left) of the sensor to +5V
// Connect pin 2 of the sensor to whatever your DHTPIN is
// Connect pin 4 (on the right) of the sensor to GROUND
// Connect a 10K resistor from pin 2 (data) to pin 1 (power) of the sensor

DHT dht(DHTPIN, DHTTYPE);

float h = 0;
float t = 0;
int val = 0;

void setup() 
{
    Serial.begin(9600);
    pinMode(DHTPIN, INPUT_PULLDOWN);
    Serial.println("DHTxx test!");

    dht.begin();
}

void loop() 
{
    delay(1000); // my addition
    
    // Reading temperature or humidity takes about 250 milliseconds!
    // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
    if (getPinMode(D2)==INPUT) {
      Serial.println(DHTPIN + " is in INPUT Mode");
      Serial.println(digitalRead(DHTPIN));
    }
      h = dht.readHumidity();
      t = dht.readTemperature();

    // Check if any reads failed and exit early (to try again).
    if (isnan(h) || isnan(t)) {
      Serial.println("Failed to read from DHT sensor!");
      return;
    }

    // check if returns are valid, if they are NaN (not a number) then something went wrong!
    if (isnan(t) || isnan(h)) 
    {
        Serial.println("Failed to read from DHT");
    } 
    else 
    {
        Serial.print("Humidity: "); 
        Serial.print(h);
        Serial.print(" %\t");
        Serial.print("Temperature: "); 
        Serial.print(t);
        Serial.println(" *C");
    }
}
```
It compiles and uploads to `readBear_Duo` no problem, but I am not getting any reading on `Serial monitor` nor on `screen /dev/ttyACM0`

Even not with:
```
#include "DHT.h"

#define DHTPIN 2    // what pin we're connected to
#define DHTTYPE DHT22   // DHT 22  (AM2302)
DHT dht(DHTPIN, DHTTYPE);

float h = 0;
float t = 0;

void setup() 
{
    Serial.begin(9600);

    dht.begin();
}

void loop() 
{
    delay(1000); // my addition

    // Reading temperature or humidity takes about 250 milliseconds!
    // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)

    h = dht.readHumidity();
    t = dht.readTemperature();

    // check if returns are valid, if they are NaN (not a number) then something went wrong!
    if (isnan(t) || isnan(h)) 
    {
        Serial.println("Failed to read from DHT");
        return;
    } 
    else 
    {
        Serial.print("Humidity: "); 
        Serial.print(h);
        Serial.print(" %\t");
        Serial.print("Temperature: "); 
        Serial.print(t);
        Serial.println(" *C");
    }
}
```
I even tried uploading it through `RBLink`...nada....

geofbaum, 2017-02-22:
Hmm it's odd that it's not even making it to the 'Failed to read from DHT' line. I'll be honest I'm a bit lost or confused on what it could be at the moment. I'm sorry to say that but at the moment I'm just sort of scratching my head as it's not something I've encountered with any of the boards or variations of the DHT sensors that I've used previously. Normally I at least get the Failed to read error, and error on compile or something like that.

There are a bunch of lines different between the DHT.h file you are using and the one that I've most recently used which is almost certainly because of you using the Seeed Grove versions of the file versus the one in this repo but just for the sake of clarity I will post what I have here for that file. I know this whole conversation got started off by the whole CPU error issue which I think is pertinent considering the non-Seeed-Grove version of the code doesn't include those CPU exclusions.
```
/* DHT library

MIT license
written by Adafruit Industries
*/
#ifndef DHT_H
#define DHT_H

#if ARDUINO >= 100
 #include "Arduino.h"
#else
 #include "WProgram.h"
#endif

// Uncomment to enable printing out nice debug messages.
//#define DHT_DEBUG

// Define where debug output will be printed.
#define DEBUG_PRINTER Serial

// Setup debug printing macros.
#ifdef DHT_DEBUG
  #define DEBUG_PRINT(...) { DEBUG_PRINTER.print(__VA_ARGS__); }
  #define DEBUG_PRINTLN(...) { DEBUG_PRINTER.println(__VA_ARGS__); }
#else
  #define DEBUG_PRINT(...) {}
  #define DEBUG_PRINTLN(...) {}
#endif

// Define types of sensors.
#define DHT11 11
#define DHT22 22
#define DHT21 21
#define AM2301 21

class DHT {
  public:
   DHT(uint8_t pin, uint8_t type, uint8_t count=6);
   void begin(void);
   float readTemperature(bool S=false, bool force=false);
   float convertCtoF(float);
   float convertFtoC(float);
   float computeHeatIndex(float temperature, float percentHumidity, bool isFahrenheit=true);
   float readHumidity(bool force=false);
   boolean read(bool force=false);

 private:
  uint8_t data[5];
  uint8_t _pin, _type;
  #ifdef __AVR
    // Use direct GPIO access on an 8-bit AVR so keep track of the port and bitmask
    // for the digital pin connected to the DHT.  Other platforms will use digitalRead.
    uint8_t _bit, _port;
  #endif
  uint32_t _lastreadtime, _maxcycles;
  bool _lastresult;

  uint32_t expectPulse(bool level);

};

class InterruptLock {
  public:
   InterruptLock() {
    noInterrupts();
   }
   ~InterruptLock() {
    interrupts();
   }

};

#endif

```

But as I said before I'm a bit confused as to why it's not working now or that you aren't even getting the 'Failed to read the sensor' error.

jadeye, 2017-08-13:
@geofbaum :+1: 
I am glad to tell you that googling `DHT.h:19:2: error: #error "CPU SPEED NOT SUPPORTED"` gets me here again.
This time, with [adafruit feather M0 BLE](https://learn.adafruit.com/adafruit-feather-m0-bluefruit-le) its working...
So for who ever gets here:
**DHT.h** file
```
// 8 MHz(ish) AVR ---------------------------------------------------------
#if (F_CPU >= 7400000UL) && (F_CPU <= 9500000UL)
#define COUNT 3
// 16 MHz(ish) AVR --------------------------------------------------------
#elif (F_CPU >= 15400000UL) && (F_CPU <= 19000000L)
#define COUNT 6
// 48 MHz(ish) adafruit feather M0 ----------------------------------------
#elif (F_CPU >= 47400000UL) && (F_CPU <= 50000000L)
#define COUNT 18
// 120 MHz(ish) AVR RedBear -----------------------------------------------
#elif (F_CPU >= 120000000L) && (F_CPU <= 130000000L )
#define COUNT 45
#else
#error "CPU SPEED NOT SUPPORTED"
#endif
```
Your calculations are good!! 
Thanx

geofbaum, 2017-09-08:
Sorry for the slow reply. Glad you got it to work, and glad we can add another device to the list that would have not been working before!

PaintYourDragon, 2019-02-14:
Try newest library - reopen if still an issue.

closed 2019-02-14 by PaintYourDragon