adafruit/DHT-sensor-library #75: Not working at all on ProMini @ 8MHz

kratochviljan, 2017-03-26:
Only change I made is the external resonator, same example code is working fine on 16MHz, after changing to 8 MHz only 0 can be read. Init pass successfully. Some issue with timing ?

houston47d, 2017-04-27:
Hello. I had the same issue on the 8 Mhz pro mini. Mine was failing periodically - was actually value dependent. Turns out the "50 us" low pulse between bits is slightly longer prior to bits 8, 16, 24, and 32, resulting in a '1' for those bits sometimes being read as a zero. Resulted in a checksum error, unless two bits were missed and then the checksum came out fine but the values were way off.
I haven't figured out how to contribute yet, but happy to send the changes to you or post them here. Basically changed the counter to uint16_t from uint32_t (the timeout comes out to 8000, so it still fits) and that increased the number of iterations, and then calculate an average value for 'lowCount' and use that for the 0/1 determination rather than the lowCount for that bit. Been working reliably since.

ehelseth, 2017-08-02:
Hi, I've also had the same issues with ProMini@8MHz and the DHT22.
I've found the same as you @houston47d, that the low pulse is slightly longer after one whole byte is sent.
Since the sample rate on the 8MHz is lower, this results in a wrong decision regarding if it's a 1 or a 0.

However, I mean that the implementation is built on a wrong assumption; the spec says that the "bits" are either ~28us or ~70us long indicating respectively 0 or 1. Also it seems that the period between each bit is NOT 50uS (only the first start bit).
But in the code the period between the bits is compared to the length of the bit, and this is wrong, but still (luckily) this mostly works.

I first did the same change as you did (using 16 counters), that enabled a few more samples (since the loop ran faster), and thus reducing the problem.
But now I've changed the code making the decision regarding if it's a 0 or a 1 to not complare the highCycles againt the lowCycles, but only look at the highCycles values.
These varies from 4/5 (for a 0) to 15/16 (for a 1). Easy to compare it against some middle value (10) to make a correct decision.
Reading your post again I see that we ended up with a similar change, but I ignore the lowCount all the way, and calculation the median based on the highCount only.

ehelseth, 2017-08-02:
Also, changed the cycles-array to be of type byte instead of uint32_t, this saves some 240 bytes of stack usage, which is a bit on a 2K MCU.

chelmi, 2017-11-03:
I've been facing the same issue, and I think the first fix (changing `expectPulse` to use `uint16_t count`) is the safest fix. Even of fast MCU it's not necessary to have 32 bits.
The rest of the fix (`highCycles > 10`) works on boards at around 8MHz, but wont work on other boards with higher frequencies. And I believe this is the reason they went with the highCycle > lowCycle solution. It's not exactly what's the spec says, but it's a safe approximation when you want to be generic. As the comment says, `expectPulse` can be used to compare the relative time of two pulses, but the actual value is hard to interpret (depends on the CPU freq. as well as the instructions in the loop in `expectPulse`).

MR: https://github.com/adafruit/DHT-sensor-library/pull/91

PaintYourDragon, 2019-01-12:
Resolved as of 1.3.1, thanks.

closed 2019-01-12 by PaintYourDragon