adafruit/DHT-sensor-library #79: DHT11 is not reading Temperature and Humidity when Arduino is powered on using DC power Supply (12V)

jyotirajsharma, 2017-07-13:
I am facing one problem. I have code which read temperature and humidity using DHT11 sensor. I uploaded following code using Arduino via USB serial, I can read values of temp, humidity. Values are being read as long as Arduino is connected to same laptop via USB.

TEMPERATURE AND HUMIDITY are being read as 0 when I power on Arduino using DC12v, 700MA adapter.

I want to deploy Arduino with DHT sensors connected with it in Greenhouse to read greenhouse environmental condition but when I power on using DC adapter or battery, it is giving "0" output.

PLEASE HELP TO SOLVE THIS PROBLEM.

DHT dht(DHTPIN, DHTTYPE);

void setup() {
  Serial.begin(9600);
  Serial.println("DHTxx test!");

  dht.begin();
}

void loop() {
  // Wait a few seconds between measurements.
  delay(2000);

  // Reading temperature or humidity takes about 250 milliseconds!
  // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)
  float t = dht.readTemperature();
  // Read temperature as Fahrenheit (isFahrenheit = true)
  float f = dht.readTemperature(true);

adams13x13, 2017-09-02:
If you power your Arduino board directly from 12v 700mA adapter, the board should end up in smoke. Probably you have 12v -> 5v converter. Grab a voltmeter and check the voltage on the POWER pin of the DHT11 sensor - as far as I remember it should work either from 3.3V or from 5V.

For sure it is not a problem of the library.

jyotirajsharma, 2017-09-02:
http://playground.arduino.cc/Learning/WhatAdapter

adams13x13, 2017-09-02:
Well, on the board you have a 12V -> 5V converter. Check which cable is connected to the "VDD" (usually pin 1) of the sensor - probably it has voltage only if powered from USB and not from the power supply. Usually it should be some "+5V", not "USB 5V" pin. Just make sure there is voltage on the "power" pin of the sensor.
By the way, from the source code of the library I would say you should get "not a number", not zero.

abhishek8503954159, 2018-06-29:
I enter a code in the Arduino and it shows the temp and humidity but after some time I again connect with power bank it shows the reading now the next day it is not showing any reading in my LCD display it is blank with square boxes left tell me what to do

PaintYourDragon, 2019-01-12:
Make sure the sensor is powered off the Arduino's 5V pin, NOT the Vin pin (which would be the unregulated 12V).

closed 2019-01-12 by PaintYourDragon