adafruit/DHT-sensor-library #83: Support for the new DHT12

nielstron, 2017-09-03:
The current version of the DHT sensor Library does not support the new DHT1 version 2.0.0, shortly the DHT12. There exists a library at [Github by WEMOS](https://github.com/wemos/WEMOS_DHT12_Arduino_Library). It would be great if it could be integrated into the existing DHT sensor library.

geofbaum, 2017-09-09:
@nielstron Is this an issue with both the Unified code as well as the older/legacy code or is it just one of them? I don't know if you've tried it with both versions of the library or not. From looking at the WEMOS code it is using the  get() command slightly differently then in the Adafruit Unified sensors library but at the same time, the biggest difference I see is that WEMOS code doesn't seem to need to pass the pin that it's attached to which even with the older WEMOS shields you still had to do. You should do a pull request to add support for the new DHT sensors.

kreintjes, 2017-09-21:
Using the DHT shield v2.0.0 would also solve my blinking issue (https://github.com/adafruit/DHT-sensor-library/issues/85), since it doesn't use pin D4 anymore.

PaintYourDragon, 2019-01-12:
Appears to be supported in current release.

closed 2019-01-12 by PaintYourDragon