adafruit/DHT-sensor-library #84: Error reading temperature / humidity

jaderarmanhi, 2017-09-08:
Dear all, Ive tried to use arduino UNO for reading temp and humidity with DHT22 sensor in my first project, and I have faced an issue. After install library as recommended here in this page, and use available code, I still got this message of the script:

"Error reading temperature!
Error reading humidity!"

In this case, what it means? Maybe library was installed incorrectly? My sensor is broken? How can I solve this?

geofbaum, 2017-09-08:
I'm assuming you are using the DHT_Unified_Sensor.ino as your code? If you are getting those errors it's just passing the errors from the code. If possible, can you test your UNO and sensor with the non-unified code to see if you get an error with that?

One last question are you using a straight DHT22/AM2302 plugged into a breadboard or are you using an imbeded sensor that is on a breakout board?

Also just to make sure, double check your wiring and compare it to the Arduino writeup just to compare.
https://create.arduino.cc/projecthub/attari/temperature-monitoring-with-dht22-arduino-15b013

geofbaum, 2017-10-01:
@jaderarmanhi Any updates or progress from your problem?

closed 2019-01-12 by PaintYourDragon