adafruit/DHT-sensor-library #85: DHT readHumidity() makes built in led blink

kreintjes, 2017-09-20:
When I call readHumidity() on the DHT object, the built in led of my Wemos D1 Mini V2 board blinks shortly. I think this should not happen, or at the least it should be possible to disable this (I looked in the source code but did not find an option for this, please correct me if I'm wrong). I am using the board with DHT sensor in my bedroom, and the blinking led is then very annoying. Any help with this would be greatly appreciated.

Gis70, 2017-09-20:
Have you any delay in your sketch ? Put a delay and the led will stop blinking during this delay maybe. Post your code.

kreintjes, 2017-09-21:
My complete sketch is kind of big, but I created a very minimum example to demonstrate the problem. I have the following code:

```
#include "DHT.h"

#define DHTPIN D4     // what pin we're connected to

// Uncomment whatever type you're using!
#define DHTTYPE DHT11   // DHT 11

// Initialize DHT sensor.
// Note that older versions of this library took an optional third parameter to
// tweak the timings for faster processors.  This parameter is no longer needed
// as the current DHT reading algorithm adjusts itself to work on faster procs.
DHT dht(DHTPIN, DHTTYPE);

void setup() {
  dht.begin();
}

void loop() {
  delay(3000);
  dht.read();
}
```

When I upload this the Wemos built-in led blinks shortly every three seconds (so when doing each read), while I expect it to be off the whole time, since I am not doing anything with the led myself. It could be that this is intended behaviour, but then I think it should be possible to turn it off.

`dht.read()` can also be `dht.readHumidity()` or `dht.readTemperature()`. The problem stays the same.

Gis70, 2017-09-21:
Ok, try this : 

`void setup() {
  pinMode(BUILTIN_LED, OUTPUT); 
}
 
void loop() {
    digitalWrite(BUILTIN_LED, LOW);   // turn off LED with voltage LOW
 
}`

kreintjes, 2017-09-21:
Unfortunately that does not work. I have found out what the problem is. I am using a [DHT shield v1.0.0](https://wiki.wemos.cc/products:retired:dht_shield_v1.0.0) which uses pin D4 as the data pin. Apparently pin D4 is mapped to the same data line as the BUILTIN_LED is, as can be seen here: https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h#L37. So therefore, every action of the DHT-sensor-library for the DHT shield will also influence the builtin led. So I am afraid there is no other solution except customizing the DHT shield to use a different pin instead.

Thanks for the help anyway @Gis70.

closed 2017-09-21 by kreintjes

Gis70, 2017-09-21:
Can't you put a delay in your main code that don't block the whole loop ? Like this, the LED will blink only when reading DHT. I guess you don't do that every 10seconds.

Good luck.

kreintjes, 2017-09-21:
Well, actually I do read the DHT every 10 seconds :p

I have found a solution though. I will solder pin 3 and 4 together on the DHT shield board, and then cut off the iron pin for pin 4. This way I can read the DHT sensor through pin 3, so it won't influence the built in led. Downside is I can't use pin 4 on the shield any more for other purposes, but I have enough pins left.

trwatson, 2020-12-04:
I have the same issue on a nodemcu and dht sensor. I don't have a shield so @kreintjes solution likely won't work for me. How can I get temperature without blinking the LED?