adafruit/DHT-sensor-library #86: Fatal error: DHT.h: No such file or directory

FlorentSms, 2017-10-01:
Good evening,

Being a beginner in the arduino, I am currently experiencing a problem with my program. I try to read the data sent by a DHT11 type sensor.

I have correctly installed the library that you put online. I verified that it's in the Arduino libraries folder and I have checked that the Arduino IDE visualizes it well.

But when I try to upload the program, that's what he tells me:
https://puu.sh/xNwjj/e0bf912c99.png
https://puu.sh/xNwn6/1ad8fc18a3.png
https://puu.sh/xNwpB/23ace6dcb7.png

I think my library is well installed but I have the impression that it doesn't recognize it.
I hope you can help me.

Good night.

Florent

geofbaum, 2017-10-01:
@FlorentSms  Can you delete the folder labeled DHT and download it through the Arduino IDE. The folder is not normally labeled just simply DHT as you show in your libraries folder. I don't think that's the issue but just to be on the safe side, I would remove the folder that you downloaded and said that you correctly installed. After you remove that folder go into Manage Libraries and search for DHT Sensor library and download it through the IDE. Try to upload it again to your Arduino after that is complete and report back if you are still having issues. The problem that you are having is normally related to something simple like it was not properly installed or that there are more than one copies of the DHT.h file floating around in your libraries folder.

Here is a picture of the Manage Libraries screen and the one you want to download to verify what you are after.
[image]

geofbaum, 2017-10-01:
Also as an aside, you aren't using the same library as this one, you are using the SimpleDHT library so if my idea fixes your problem that is possibly a start. The SimpleDHT code is hosted at the following link I believe, https://github.com/winlinvip/SimpleDHT

FlorentSms, 2017-10-03:
Hello @geofbaum ,
I have try your solutions but i have, for the 2 solutions, the same problem.

Thanks in advance.

Florent.

geofbaum, 2017-10-03:
@FlorentSms Can you upload an image of what the files in your DHT library folder look like. I just want to check to make sure capitalization or if somehow the files are misnamed somehow.

FlorentSms, 2017-10-03:
@geofbaum 
Yes no problem:
https://puu.sh/xPbas/82a0051685.png
https://puu.sh/xPbbC/2a5873b271.png

SuriGangs, 2017-12-01:
Hi, could you solve your problem?

JackZ0, 2018-12-27:
\Adafruit_AM2320_sensor_library/Adafruit_AM2320.h:41:29: fatal error: Adafruit_Sensor.h: No such file or directory

closed 2019-01-12 by PaintYourDragon

isekaimaou1109, 2020-04-06:
> @FlorentSms Can you delete the folder labeled DHT and download it through the Arduino IDE. The folder is not normally labeled just simply DHT as you show in your libraries folder. I don't think [...]

yeah i see and when i do it as you said, i also get the same error
delete DHT folder in libraries then install it via Arduino IDE and still get that error and don't have any file DHT.h floating around my sketch
help me please, graceful and tks so much <3

Diohicham, 2020-11-11:
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Board: "Arduino Uno"

sketch_nov10a:1:10: fatal error: dht.h: No such file or directory

 #include "dht.h"

          ^~~~~~~

compilation terminated.

exit status 1

dht.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

ZsoZ, 2021-09-02:
#include <DHT.h>

Arely-M, 2022-06-19:
Puede ser porque falta una librería  Adafruit Unified Sensor, esta viene al instalar todo desde el administrador de bibliotecas
[image]

hernandez2david, 2022-07-10:
> Puede ser porque falta una librería Adafruit Unified Sensor, esta viene al instalar todo desde el administrador de bibliotecas
You Rock Arely! This worked

selenium78965, 2025-06-10:
It's working!!Thanks

> Puede ser porque falta una librería Adafruit Unified Sensor, esta viene al instalar todo desde el administrador de bibliotecas [image]