adafruit/DHT-sensor-library #98: DHT11 is NOT able to read data with Arduino Pro Mini, 3.3V 8MHz.

jyotirajsharma, 2018-03-05:
I am using DHT11 and Arduino Pro Mini, 3.3V 8MHz device with no USB. I am programming with FTDI adapter. It is successfully programmed after I set the connection settings between FTDI and Arduino Pro Mini, as shown below.

But when I try to read data from DHT11 sensor, it is giving erros; sometimes "connect error" and sometimes "Ack error. I even tried supplying VCC power supply(3.3V or 5V) and GND to DHT11 pins externally from Arduino Uno R3, but it is still not able to read data from the sensor. The same code is working fine Arduino UNO R3.

Please let me know if I am missing anything because I am new to Pro Mini but have been using Arduino UNO R3.

```
**Settings used while programming Arduino Pro Mini:**
Processor: 8 MHz (3.3V, 8 MHz)
Programmer: USBasp

**FTDI                 Arduino Pro Mini**
  DTR       <--->      DTR
  VCC       <--->      VCC
  GND       <--->      GND
  TX        <--->      RX
  RX        <--->      TX
  DHT11 (data pin)<----> Digital Pin 5
```

chelmi, 2018-03-05:
See https://github.com/adafruit/DHT-sensor-library/issues/75

jyotirajsharma, 2018-03-05:
As per this post, it seems latest version of DHT11 library fixes this issue. I am using the latest version, but this problem seems NOT fixed. Please confirm.

letsbyteit, 2018-11-11:
I have the same issue with DHT22 and the 8MHz Arduino Pro Mini. I also tried several other libraries out there, none work. I tried different Pro Minis, I tried different DHT22, I confirmed the DHT22 works correctly with a Raspberry Pi. I tried using an external 3,3V power supply and it still does not work. I've tried the suggestion from #75 https://github.com/adafruit/DHT-sensor-library/issues/75#issuecomment-341613008

I've also tried the proposed PR https://github.com/adafruit/DHT-sensor-library/pull/91

Any hints, ideas or anything? Only chance to switch to the 16MHz or what do you suggest?

fisenkodv, 2018-11-11:
See working example here https://github.com/fisenkodv/home-temperature-monitoring/blob/master/src/Monitoring.Sender/Monitoring.Sender.ino

See details here https://blog.fisenko.page/home-temperature-monitoring/

letsbyteit, 2018-11-11:
@fisenkodv That's a really nice blog-post. Unfortunately, with the Tillaart-Lib I'm always getting the `DHTLIB_ERROR_ACK_L` error. I used your sample code with the provided version of the Tillaart-Lib.

fisenkodv, 2018-11-11:
@letsbyteit have you tried the same [DHT22](https://amzn.to/2Jnt5lg)? Because it works fine in my case with Arduino Pro Mini 3V. Could you try the same code as here https://github.com/fisenkodv/home-temperature-monitoring/tree/master/src/Monitoring.Sender? Just curious.

letsbyteit, 2018-11-13:
@fisenkodv  I have the same chip DHT22 AM2302, although mine does not have that 4 -> 3 pin adapter board attached. So, I've got to ignore that third pin from the left. But the sensors itself are working with my Raspberry Pi on 3,3V.
I've tried your code and I'm still getting the same `DHTLIB_ERROR_ACK_L` error as always with the  Tillaart-Lib.
I'm really clueless with this, there are many tutorials out there using the same setup but for me, it does not work.

vsch, 2018-12-28:
I just got the DHT22 working with Pro Mini 3.3v 8MHz. I was getting a lot of checksum errors.

I added raw count trace option to the library and saw that often the low count == high count. In this case the library treats it as a 0 bit, but real 0 bits have low < high by a margin, while 1 bits have high > low with a much lower margin.

The count trace showed that DHT22 has a longer low pulse for the first transmitted bit of bytes 1, 2 and 3 which makes the problem worse because the low pulse is extended by a count 2-3 while the high pulse is not, making the high == low condition result more often for these bits. Exactly what I was seeing in the trace without raw counts. High bit would flip to 0, most often causing a checksum failure.

Changing the code from `if (highCycles > lowCycles) {` to `if (highCycles >= lowCycles) {` solved the problem almost all of the time. I would even go as far as changing the condition to `if (highCycles+1 >= lowCycles) {` for at least DHT22 because the 0 bit margin of difference is big enough to handle a couple of extra counts, while the 1 bits are very close.  There is a comment mentioning that high time equal to low time for weird cases but does not mention the sensor so I can't be certain if this test needs to take the sensor into account.

PaintYourDragon, 2019-01-12:
Resolved as of 1.3.1, thanks.

closed 2019-01-12 by PaintYourDragon