adafruit/DHT-sensor-library #100: dht issue

krishna1703, 2018-03-18:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

krishna1703, 2018-03-18:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Uno"

fhfhfh:11: error: 'dht' does not name a type

 dht DHT;

 ^

C:\Users\krishna\Documents\Arduino\fhfhfh\fhfhfh.ino: In function 'void loop()':

fhfhfh:55: error: expected unqualified-id before '.' token

   DHT.read11(dht_dpin);

      ^

fhfhfh:58: error: expected primary-expression before '.' token

   humi=DHT.humidity;

           ^

fhfhfh:63: error: expected primary-expression before '.' token

   tem=DHT.temperature;

          ^

exit status 1
'dht' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

i got this type of error please help me to solve this error

dhalbert, 2018-03-18:
`DHT` (all capitals) is the name of the class. So do `DHT dht;` instead of `dht DHT;`. For example: https://github.com/adafruit/DHT-sensor-library/blob/c97897771807613d456b318236e18a04b013410b/examples/DHTtester/DHTtester.ino#L24

This is a support issue, not a library bug, so next time please ask in https://forums.support.com or in discord: https://adafru.it/discord. Thanks.

closed 2018-03-18 by dhalbert