adafruit/DHT-sensor-library #106: ESP8266 and DHT22 frequent hangs

HumamHelfawi, 2018-05-05:
- Arduino board:  ESP8266 (NodeMCU)
- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.4**

I have DHT22 sensor attached to an ESP8266. When running a sketch that read the sensor values, it suffer from frequent hangs (something like each 1 second). 

All outputs are stopped (Sound from speaker or even the Serial output) during those hangs.

Here is a small Sketch to reproduce:

```
    #include "DHT.h"
    DHT dht(D1,DHT22);
    void setup(){
      Serial.begin(115200);
    }    
    void loop(){
      Serial.println(dht.readTemperature(true));
      Serial.println(dht.readHumidity());
      delay(1);
    }
```

P.S. it is not about the small `delay(1)`. Even if I read the sensor values each 1 second, this problem will occur. To make sure, you may run this example:

```
    #include "DHT.h"
    DHT dht(D1,DHT22);
    
    void setup(){
      Serial.begin(115200);
    }
    
    unsigned long last = 0UL;
    void loop(){
      if(millis() - last > 1000){
          Serial.println("#########" + String(dht.readTemperature(true)));
          Serial.println("#########" + String(dht.readHumidity()));
          last = millis();
      }
      Serial.println(String(rand()));
      delay(10);
    }
```
After flashing it, opens serial monitor and you can see clearly the hangs (I put random number printing so your eyes can not cheat you by an cashed old value;) )

HumamHelfawi, 2018-05-05:
Sorry forget it while copying. It is exist in the setup function

HumamHelfawi, 2018-05-07:
Thank you very much!

>  I'm still not sure what you mean by, All outputs are stopped (Sound from speaker or even the Serial output) during those hangs.
For example there is an mp3 file that is being played using the ESP8266Audio library which uses non-blocking buffering for music playing. During those hangs, the output stops. The blocking time is too short, so I do not think that the buffer has finished and there is nothing to be played. I think the serial port (which is the speaker attached too) has been blocked in some way.

PaintYourDragon, 2019-02-14:
Try newest library - reopen if still an issue.

closed 2019-02-14 by PaintYourDragon