adafruit/DHT-sensor-library #107: DHT22 Failed to read from DHT sensor!

ahmedawais, 2018-05-25:
Hi 
     i have connected the data pin with IO17 of ESP32 WROOM board and changed the pin in code as 17 while using 10kohm resistance working with Arduino IDE initallly serial monitor showed some temperature and humidity readings but in the next run of sketch serial monitor shows Failed to read from DHT sensor!
does anyone know the reason ?

Jan2220, 2018-05-31:
I have the same issue with the dht22 connected to my raspberrypi 3b using a 10kohm resistor. It worked for a while but after around 1-2 hours without really doing anything it stopped working and the example script in the library just exports `Failed to get reading. Try again!`...

tvwerkhoven, 2018-09-23:
I had a similar problem, but it disappeared after ±40 seconds. "Failed to read" therefore could also mean "Initializing". Hope this helps. I'm using Wemos D1 mini pro v2.0.0 with a DHT22 module.