adafruit/DHT-sensor-library #111: readHumidity(bool force) `force` is not used in the code

FeyverkSoft, 2018-07-26:
https://github.com/adafruit/DHT-sensor-library/blob/c97897771807613d456b318236e18a04b013410b/DHT.cpp#L72

`force` is not used in the code. :)

Tijgerd, 2018-08-16:
I think this boolean is still present to support backwards compatibility.

L0laapk3, 2019-01-10:
No, its definitely being used in the read function

Tijgerd, 2019-01-10:
Please show me in the code below where the _force_ bool is being used
```C++
float DHT::readHumidity(bool force) {
  float f = NAN;
  if (read()) {
    switch (_type) {
    case DHT11:
      f = data[0];
      break;
    case DHT22:
    case DHT21:
      f = data[0];
      f *= 256;
      f += data[1];
      f *= 0.1;
      break;
    }
  }
  return f;
}
```

L0laapk3, 2019-01-10:
Its fairly obvious if you compare `readTemperature` to `readHumidity`:
https://github.com/adafruit/DHT-sensor-library/blob/c97897771807613d456b318236e18a04b013410b/DHT.cpp#L72-L74

https://github.com/adafruit/DHT-sensor-library/blob/c97897771807613d456b318236e18a04b013410b/DHT.cpp#L35-L38

They're both supposed to call `read(bool force)`
I think its fairly obvious.

https://github.com/adafruit/DHT-sensor-library/blob/c97897771807613d456b318236e18a04b013410b/DHT.cpp#L123-L129

Tijgerd, 2019-01-10:
I understand what you are saying but `force `is NEVER used in the readHumidity function!
It defaults to `read(false)`

If you use the readHumidity function, you can give whatever value you want to `force` but it won't have any impact!

https://github.com/adafruit/DHT-sensor-library/blob/c97897771807613d456b318236e18a04b013410b/DHT.h#L47

L0laapk3, 2019-01-10:
Im not claiming the current code isnt syntactically correct, it obviously is or it wouldn't have compiled and they would have never published it.

The whole point of this issue is that if you call `readHumidity` with `force = true`, you should probably actually force an update of the sensor values (analogue to `readTemperature`) instead of just doing nothing with it.

Tijgerd, 2019-01-10:
Tell me then, where is the `force` bool transferred into the `read()` function?
When you just call `read()` without any value between the hooks " ( ) ", the default `false` value will be used in the `read()` function.

e.g.: calling `readHumidity(true)` will not force your MCU into doing anything different in comparison to calling `readHumidity(false)`

prove me wrong

L0laapk3, 2019-01-10:
Thats the entire issue mate, `force` doesn't do anything right now and it should. I don't know how many other ways I need to reformulate the same thing so I'm just going to give up

Tijgerd, 2019-01-10:
I believe we are both saying the same thing in a different way.
We both agree that the function doesn't do anything with the provided bool, but it should.

PaintYourDragon, 2019-01-12:
Fixed in 1.3.1. Thanks!

closed 2019-01-12 by PaintYourDragon