adafruit/DHT-sensor-library #116: Reopening issue #94 'ESP8266 and DHT22: NAN errors using libraries >= 1.2.0'

shaunakvyas1, 2018-11-29:
I kindly request to reopen the issue #94 "ESP8266 and DHT22: NAN errors using libraries >= 1.2.0" as I do not find it is resolved yet and I have a justified reason for it. I am working on a Wemos D1 R2 board's interfacing with DHT22 sensor and as usual I have been getting the NaN errors using the given libraries. I went through all the given solutions given for issue #94 individually as well as combining, but no solution is working out for me. My Wemos D1 R2 board has nothing wrong as I have already accomplished a Data Acquisition with IoT project on it. I am not kind of a person who asks random questions. I try to solve the issue myself first. If I really find myself stuck up, I ask a question. I am being advised in issue #94 to go with other forums, but actually I would like to inform that this issue is studied, analyzed and approached the best way compared to other forums. Github has taught me never to give up, so I kindly request to reopen the issue #94 as I am the one who is still to be answered. 

Shaunak Agastya Vyas

SergiuToporjinschi, 2018-12-14:
I think I have a similar issue. I'm using an ESP8266-01 and DHT22, the esp should send the values over mqtt; After one or two days is sending only the fail of DHT reading message. 
It only works if I power off the entire circuit and plug it back.

geofbaum, 2018-12-22:
@SergiuToporjinschi please see if the information at the following link helps
https://github.com/RobTillaart/Arduino/issues/102

@andrewlalis are you only seeing the error messages from the NaN values being returned or are you seeing some valid readings interspersed with valid ones. Also have you done any of the suggested fixes from #94 ? All of these problems especially with the ESP8266 or Wemos boards probably stem back to something inherent with those boards like a clockspeed and read timing issue especially with sensors like the DHTs which are well known to be slow and very particular about how and when you read them.

Kentric, 2019-01-12:
Same problem here. Wemos D1 R2 + DHT22. The nan errors come intermittently. I am taking readings every 5mins and some times they are fine. Fortunately my readings are not critical so I ignore the errors and continue. Seems like the nan error is mostly from the humidity sensor. I have tried everything recommended to eliminate these errors but nothing helps.

Jaipharma, 2019-01-15:
last library version 1.3.1 is still has problem with NAN error .
because of sensor signals are vary too much from data sheet . 
sensor response too fast than define in data sheet.
As I inspected in raw signal.
first low pulse from sensor come is 23-55 us
duration of first low pulse is 13-38 us (80 us (75-85) in data sheet)
seconds high pulse  duration is about 35-73 us (80 us (75-85) in data sheet)

look! sometime first low signal come in 23 us and duration of pulse is 13 us.
if you use delayMicroseconds(60);  23 + 13 = 36 us , you will step over first low pulse from sensor.
then NAN reading occur.

I have modified few codes in lastest libray 1.3.1 to fix this problem like here.

  uint32_t cycles[80];
  {
	//**modified by jaipharma**//
	// Turn off interrupts temporarily because the next sections
	// are timing critical and we don't want any interruptions.
	InterruptLock lock;

    // End the start signal by setting data line high for 40 microseconds.
    pinMode(_pin, INPUT_PULLUP);

    // Now start reading the data line to get the value from the DHT sensor.
    //delayMicroseconds(60);  // Delay a bit to let sensor pull data line low.
	//don't use delay
	//waiting for start signal low pulse.
	if (expectPulse(HIGH) == TIMEOUT) {
		DEBUG_PRINTLN(F("DHT timeout waiting for start signal low pulse."));
		_lastresult = false;
		return _lastresult;
	}
	//**modified by jaipharma**//

    // First expect a low signal for ~80 microseconds followed by a high signal
    // for ~80 microseconds again.
    if (expectPulse(LOW) == TIMEOUT) {
      DEBUG_PRINTLN(F("DHT timeout waiting for start signal low pulse."));
      _lastresult = false;
      return _lastresult;
    }
    if (expectPulse(HIGH) == TIMEOUT) {
      DEBUG_PRINTLN(F("DHT timeout waiting for start signal high pulse."));
      _lastresult = false;
      return _lastresult;
    }
.............................

full code is here  https://github.com/jaicurx/DHT-sensor-library-jai

it works fine for me but sometime sensor will stop working (signal always high)
fixed by turn power off , wait a few seconds  then turn power on again.
sensor will start working again.
I think DHT22 is not good enough, unstable sensor. try to use SHT3x instead.

ladyada, 2019-01-17:
are any of the people in this threading using both Adafruit ESP8266 board and Adafruit DHT sensors?

khjoen, 2019-01-18:
PR #70 has this kind of trick.

khjoen, 2019-01-18:
> //**modified by jaipharma**//
> // Turn off interrupts temporarily because the next sections
> // are timing critical and we don't want any interruptions.
> InterruptLock lock;
>
> ```
> // End the start [...]

PR #70 has this kind of trick.

mrzino, 2019-01-26:
I also had lot of issues with DHT22 and ESP8266-01. Used a 5.1k Pullup resistor. I was only getting readings from the sensor every 20th try, else just getting NAN.
I also tried the code-changes suggested by [jaipharma](https://github.com/adafruit/DHT-sensor-library/issues/116#issuecomment-454261351), but that also didn't help.

I am using the same ESP8266-01 with DHT22 also with [nodemcu-firmware](https://github.com/nodemcu/nodemcu-firmware), and there it worked nicely. Therefore I have compared their implementation with the one in this library and made few changes. Now it works nicely for me (haven't seen NAN since).

`
uint32_t cycles[80];
  {
    // Turn off interrupts temporarily because the next sections
    // are timing critical and we don't want any interruptions.
    InterruptLock lock;

    // End the start signal by setting data line high for 40 microseconds.
	digitalWrite(_pin, HIGH);
	delayMicroseconds(40);
    pinMode(_pin, INPUT);
	
    if (expectPulse(LOW) == TIMEOUT) {
      DEBUG_PRINTLN(F("DHT timeout waiting for start signal high pulse."));
      _lastresult = false;
      return _lastresult;
    }
    if (expectPulse(HIGH) == TIMEOUT) {
      DEBUG_PRINTLN(F("DHT timeout waiting for start signal low pulse."));
      _lastresult = false;
      return _lastresult;
    } `

I think there is a error in the sensor data-sheet. I says, that the sensor will pull up bus's voltage within 20-40us, but I think the sensor expects the MCU to do that (also also shown by the color in the chart).

Hope this helps.

JmBelisle, 2019-02-11:
I second this request but also propose a fix.
**_Summary_**
The code of V1.3.2 occasionnally misses the first LOW DHT signal of the 80us LOW/80us HIGH signal pair that preceeds the 40 bit information data stream.  This is because occasionnally, the DHT takes more that 40us (line 174) to pull the line down and thus the DHT.cpp code expecting a LOW (line 182) but reading the trailing edge of the priorly ESP generated HIGH (line 187), gets out of sync with the bit stream and interprets the trailing edge as the HIGH signal of the 80us/80us pair. When the DHT finally sends the 80usLOW/80usHIGH pair, the code incorrectly interprets it as the first data bit, thus adding a bit and throwing the ESP computed checksum out of balance with the DHT computed checksum.

**_The fix (tested on ESP8266 NodeMCU)_**
Increase the wait time before reception of the 80usLOW/80usHIGH start signal from the DHT from 40us to 50us (minimum) at line 174 in DHT.cpp.

**_Supporting evidence_** 
Using V1.3.2 of the library with NodeMCU ESP8266 and Adafruit DHT22 modules.
I got 4%-8% of calls returning NAN. By going to debug mode (#define DHT_DEBUG) in dht.cpp I found that ALL errors were of type "DHT checksum failure" such as below (_Italics_ : my program; **Bold** produced by dht debug mode) :
      _JMB : Read# 000087_
      **Received from DHT:**              
      **80, C0, 80, 67, 28 =? 27**
      **DHT checksum failure!**
      _JMB : Read# 000088_
      **Received from DHT:
      1, 81, 0, CE, 50 =? 50**
Looking at the 40 bit pattern of the bytes received in these 2 consecutive calls (3 seconds apart), I realized they were the same but shifted by one bit: The correct call bit pattern was the same as the Checksum failure one but shifted one bit left. As below:
**80,C0,80,67,28** == 10000000,11000000,10000000,01100111,00101000
**01,81,00,CE,50** == 00000001,10000001,00000000,11001110,01010000

From these observations, I surmised that the DHT22 had transmitted the right information in both cases but that somehow a leading "additional bit" had been read ahead of the correct binary sequence thus explaining the shift. Not having a scope to look at the bit stream in realtime, I modified the DHT.cpp library to :
1. capture the (duration) of the two LOW and HIGH level data pin signals (from the DHT22) that preceed the 40 bit data stream (measured in lines 182 and 187 of DHT.cpp). 
2. print the duration of the 80 LOW and HIGH level data content that provide the 40 bits of data (captured in array "cycles" in lines 202 and 203).  The output became : 

Received from DHT:
80, C0, 80, 67, 28 =? 27
DHT checksum failure!
**000 009**   correspond to 80usLOW/80usHIGH start signal pair from DHT
**118**   80  83  82  83  83  82  83  83 103   82  83 ... LOW signal of bit signal pair 
**123**   39  40  40  39  40  40  39 113 115  40  39 ... HIGH signal of bit signal pair
    1     0    0    0    0    0    0    0     1     1    0    0 ... Resulting bit
Received from DHT:
01, 81, 00, CE, 50 =? 50
**090 121**
  75   79   83   83   83   83   83   83 103   83  83 ... LOW signal of bit signal pair
  40   40   39   39   40   39   39 113 115   39  39 ... HIGH signal of bit signal pair
    0     0     0     0     0     0     0     1     1     0    0 ... Resulting bit

One reads/interprets the info as follows:
**000** is the calculated duration (by DHT.cpp) of the 80usLOW part of the start signal pair
**009** is the calculated duration (by DHT.cpp) of the 80usHIGH part of the start signal pair
**118** is the calculated duration (by DHT.cpp) of the LOW part of the LOW/HIGH signal for the first bit 
**123** is the calculated duration (by DHT.cpp) of the HIGHpart of the LOW/HIGH signal for the first bit 
       (and given that 123 > 118) that bit is a "1".)
Note that from the read that did not generate a checksum error we see:
**090** is the calculated duration (by DHT.cpp) of the 80usLOW part of the start signal pair
**121** is the calculated duration (by DHT.cpp) of the 80usHIGH part of the start signal pair
Also note that 90us and 121us are larger than the 80us spec.... but this information is NOT generated from a scope so it's accuracy is not total.  Suffice it to say 121 is still 50% larger than 80 and thus specs are not quite met.

Likely explanation: say the DHT pulled the line low 49us (and not 40us) after ESP relinquishes control after going to input mode; then after a 40us delay (line 174)  when the ESP expects the LOW part of the start pair, the line is still HIGH from the ESP command at line 171.  The code thus registers a zero LOW duration (because the line is HIGH, not low).  The duration of the HIGH portion is 9us, which corresponds to difference in time between the 40us wait on part of the ESP and the 49us of actual time the DHT took to take the line down.  The DHT.cpp code then begins the information bit read loop (line 201) and mistakes the now incoming start signal LOW/HIGH pair as the first bit... thus adding a bit and shifting the bit stream as we observe in the printed Debug data..

**_Conclusion_**
The fix is simply to increase the 40us delay to 50us, assuming that the 9us delay for the DHT in starting the LOW/HIGH start signal is the maximum time the DHT will ever go over the 40us spec (no certainty here). 
 I did that modification and ran 10000 successive reads on each of 3 DHT sensors from ADAFRUIT and got ZERO errors over those 30000 reads.  I ran the same tests with a delay of 55us and got the same results of ZERO errors. 

I agree that lacking scope and a larger sample set of sensors, the 55us is a better choice of delay. 
 I also agree that because these errors are timing related, that the fix may not work for other platforms since I have not performed such tests.  I intend to forward my proposed changes in both the debug capture and printouts, and the timing delay increase through a pull request.

jmb

khjoen, 2019-02-12:
@PaintYourDragon and @ladyada 
Please consider having a look at the line 154 to 182 of ba8cd46debe1cb855fd2b46e8a9b6cf671673abc in PR #70 that are mostly comments.

If @JmBelisle is right, this PR hold a solution variable waiting times from one sensor to another and/or  times for the each MCU to pull the line LOW.   Solution may not be ideal, but I plan on reviewing once I receive the ESP01 I ordered so I can test.   

Datasheet is clear about the time to wait for a response from the sensor after the start signal is variable.  Thus, no constant will work for all.

PaintYourDragon, 2019-02-14:
1.3.3 changes the pullup time to 55 microseconds by default, which you can optionally override by passing a different value to begin(). Also added yield() immediately before and after the interrupt-disable period.

@khjoen No plans to merge PR #70 as written because the loop count will be highly variable by architecture (requiring a huge collection of #ifdefs to handle, and may change with compiler and core library updates without or knowledge). IIRC micros() typically has a 4 usec resolution which should be sufficient. But it all seems kind of moot since one of the two expectPulse() calls that follow should catch a timeout themselves anyway.

PaintYourDragon, 2019-02-14:
Is anyone experiencing these problems NOT using an ESP8266?

JmBelisle, 2019-02-14:
**Warning regarding version 1.3.3** : the introduction of yield() in the loop to capture LOW/HIGH data cycles now generates NANs which, with  "DHT_DEBUG" defined, come out as "DHT timeout waiting for pulse".
Removing the two references to yield() (in the DHT.h file) solves this (we get zero NANS).
If yield() is necessary for the ESP8266 background tasks to properly work, then we have a serious issue. 
However, quick measurements show that the LOW signal count capture averages a total of 3430us and the HIGH signal count capture averages 3100 (more variable because dependant of data bits values (1s take more time than 0s).  Say a total of 7000us.  If the background processes suffer if not granted cycles for 7000us, then we have the issue.  If not, then I suggest removing the yield() statements  from DHT.h of V1.3.3.

JmBelisle, 2019-02-14:
@PaintYourDragon : I will test 1.3.3 on other chips I have and come back in a comment in this stream.
And thanks for the prompt revision of 1.3.2 : much appreciated.

ladyada, 2019-02-14:
ESP8266 cannot guarantee that it will not be interrupted and not crash unless we yield() beforehand

PaintYourDragon, 2019-02-18:
1.3.4 moves the yield() out of InterruptLock and handles it a little earlier (but still ample time, watchdog isn't likely to trigger during DHT read even in the worst case).

kreuzundquer, 2019-03-05:
Im having the same issue in 1.3.4

Board used: NodeMCU v3.2

With DHT_DEBUG defined i get "DHT checksum failure!".
I tried different timings (40 - 255us) in the begin function, but nothing worked.
It looks like im having the same issue as @JmBelisle in his long and detailed post from the eleventh of feb with a leading high bit which shouldnt be there but i cant get rid of it. Shifting everything by 1 bit gives me reasonable results in reading outputs. This is the only solution that works for me so far.

JmBelisle, 2019-03-05:
Hi,
In my case, 1.3.4 has solved the past issues with zero NaNs on NodeMcu, and tested against 4 different DHT22 sensors overs 10s of thousands of readings.
A couple questions:
1) Are you getting these results systematically?
2) Are these occurrences significant (what percentage of readings?)
3) is your DHT22 bare (no pull-up on dataline) or an Adafruit sensor (that comes built in with such a resistor)?
4) Are you waiting more than 2 seconds between readings?
Regards
JmBelisle

kreuzundquer, 2019-03-06:
Apparently I was to dumb to use the library correctly. You know that feeling, when you are overthinking stuff and don't see the smallest mistake? Thats basically what happened to me and that's why I already deleted my comment from yesterday. I didn't use the begin method in my code and got the leading bit issue. I compared the Datasheet Signal Flow and the librarycode and noticed that something is missing. Thats when I stumbled over the begin method and saw that I didn't used it in my code. Yeah, so basically it was my fault, but thanks for your fast reply @JmBelisle

roXDesign, 2019-03-08:
I do have a similar issue - I try reading the temperature and humidity values every 5 seconds (which should be fine?) and after power up and initialization a DHT22 will always return NaN values, while DHT11 Sensors work just fine. If I unplug the DHT22 Sensor in powered mode and plug it in again, it will suddenly start to work and never return NaN values ever again... which makes it even more confusing to me.

One difference - since I am loading the Port Configuration from the Web, I am using Pointers to access the Objects. Can this be a problem? 

`
void initDHT4() {
 DHT_Unified* dht4 = NULL;
 dht4 = new DHT_Unified(DHTPORT4, DHTTYPE22);
 dht4->begin();
 dht = dht4;
}

float readDHTHumidity() {

     sensors_event_t event;
     float humidity;
     dht->humidity().getEvent(&event);

        if (isnan(event.relative_humidity)) {
          Serial.println("NAN Humidity!");
        return null;
        }
        else {
          Serial.print(F("Humidity: "));
          Serial.print(event.relative_humidity);
          humidity = event.relative_humidity;
          Serial.println(F("%"));
          return humidity;
        }
    }
}`

Misiu, 2019-03-23:
Hi all.
I have a similar issue.
I use Wemos D1 and DHT22 Shield.
In my case the problem was the producer of chip - I have ASAIR sensor, not AESONG. I don't know why but it doesn't work. I'm going to order a different sensor and we will see if it works.

sjdttt, 2019-04-06:
I am having exactly the same problem with WEMOS Mini D1 and DHT22. I have found that the AESONG variant works fine and the ASAIR does not (using "Arduino" Pin 2 (D4) with/without pullup (10k builtin)..... hot plugging the AESONG one, after WEMOS initialisation  starts it working.......

Misiu, 2019-04-06:
@sjdttt try changing pin from D4 to other. You must cut trace on PCB and solder extra wire. You can try jumper wires first. You must connect GND, +3.3V and D5, then change in code.
I've already ordered sensors from other seller, hopefully they won't have such problems.

sjdttt, 2019-04-06:
@Misiu ... hard to change pins for this project as I have used every pin on the Wemos, and the pin assignment has already been very difficult. 

Of course my breadboard version used the AESONG and I had no problems..... I made PCB's (4)  for the live system and bought new DHT22s which turned out to be ASAIR, and then my problems started.

 I have swapped out AESONG DHT22s from another project (which uses a different pin and works ok with ASAIR and now have my new  project working.

Using pins which stay steady during the boot/reboot gives working ASAIR DHT22s..... I will revisit the pinouts on any future version, but am more likely to switch to an ESP32 and use BME280 (or another alternative....)

Note:- My initial post should have said ... hot plugging the ASAIR started it working.

AnakinLinux, 2019-04-18:
Howdy, everybody, I am also having the same issue as everybody else and I have tried every single possible fix that has been posted on here and every other thread I can find and none have worked. I'm at my wits end here I CANNOT find a solution.

Misiu, 2019-04-18:
@AnakinLinux try this solution:
> try changing pin from D4 to other. You must cut trace on PCB and solder extra wire. You can try jumper wires first. You must connect GND, +3.3V and D5, then change in code.
> I've already ordered sensors from other seller, hopefully they won't have such problems.

LizardWizzard, 2019-06-02:
Faced same problem with ESP8266_WEMOS_D1R1 and ASAIR AM2302 (DHT22). Turning on debug mode gives me

>  DHT timeout waiting for start signal low pulse.

using DHTtester sketch from examples folder.

Switched to another library: https://github.com/beegee-tokyo/DHTesp and it worked for me

CyanoFresh, 2019-06-07:
Same on Wemos d1 mini + DHT22, D4 pin is used

Lordofrealms, 2019-06-22:
This should fix this issue on DHT11s for interrupt enabled MCs: https://github.com/Lordofrealms/DHT11LUM

Misiu, 2019-06-24:
@Lordofrealms maybe You could create PR with a fix for this problem?

roXDesign, 2019-06-24:
> Switched to another library: https://github.com/beegee-tokyo/DHTesp and it worked for me
> UPD:
> Tried PIN d4 instead of d2 and it worked even with this library. Taking my words back :) But I think it should be noted in docs (or in the tutorial).
Hmm.. doesnt work for me :(

ahmadSum1, 2019-07-04:
I am having this weird problem with DHT22 and esp8266(NodeMCU).
It works fine with continuous reading and starts generating **_nan_** after random intervals(sometimes after couple of hours, sometimes after several days) and the nans continue until a power recycle is performed. Restarting the ESP from (code/button) doesn't work. But unplugging and then replugging the DHT22 works. 
So seems like the DHT22 is freezing.

Lordofrealms, 2019-07-04:
I would be happy to look into it if I had an esp and a dht22. Have you
tried the dht22 on another mc?

ahmadSum1, 2019-07-05:
> I would be happy to look into it if I had an esp and a dht22. Have you tried the dht22 on another mc?

yes and no. The library normally works fine with arduino(uno/mega), ESP32, esp8266.
But the DHT22 freezing issue only occurred on this particular project with ESP8266 where the sensor is running continuously for days on a remote location. As I mentioned it occurs after a random interval, I couldn't replicate the incident at my lab.

Lordofrealms, 2019-07-05:
I would put forth three possible explanations.

1. Bad hardware as you have asserted. Unlikely but possible. (Try another
mc and dht22 to eliminate this possibility).

2. Power sag browning out either the mc or the dht22. (Put the mc on a ups
to eliminate this possibility... if you're not using a premade mc this
could also potentially be incorrect cap sizing)

3. Time rollover not properly coded in the library. I didn't dig into the
adafruit library I just gave up and wrote my own. Not sure if possibly they
have a millis or micros rollover error. Comparing the two timestamps vs
subtracting them will make the library break after the time rolls over.
However, this would be happening at a regular time. You can Google millis
rollover arduino and should be able to find the time this occurs. I don't
have it memorized. Once you know the time you can see if it is happening at
a regular interval. You could also just constantly poll your device and log
it to serial and see if it keeps stopping at the same time. If it is the
library you either need to switch libraries or try to fix it.

Good luck! Let me know what you find.

ahmadSum1, 2019-07-05:
Unless it's a bad batch, 1. is not the issue. I've tried all the four combinations.
2. can't also be the case since one of the devices was actually powered off of a UPS. And God, I wish it wasn't. As load-shedding didn't affect it, once it was starting giving _nan_ the only way to bring it back was to go up to that location and disconnect and reconnect the sensor by hand (only to repeat the same after several hours/days).
And for 3. shouldn't rebooting the device fix that? As I mentioned, I'am able to restart the devices remotely using the esp8266's restart function. That doesn't fix the nan issue unless its a power recycle.
I am starting to think its more of a DHT22 hardware issue than the library itself and a fair warning should also be provided with the library about the DHT22 sensor's/library's unreliability.

Lordofrealms, 2019-07-05:
Sorry sounds like you've tried it all. The dht22 is highly timing dependent
since it uses a single data pin that the mc and dht22 alternately control.
I don't have the esp8266 or a dht22 to test on. Even then if you don't have
the issue in your lab I might not be able to duplicate the problem. If you
don't have the issue in your lab you could try making your lab setup more
like your field setup.

Only other thing you could do would be put a resistor in series with the
dht22s 5v that current limits it to like 50ma then put a relay to ground on
the 5v pin too. Then should be able to trigger the relay to reset the
dht22. Its a kludge but it should work.

Hope that helps.

ahmadSum1, 2019-07-16:
I was able to recreate the nan issue in the lab.
This time I was using about 2 meters long ribbon cable for connecting the DHT22.
1. DHT22 vcc=3.3v yields higher frequency(every 5~8 hours) of **_nan_** (DHT22 hang) issue. 
2. DHT22 vcc=(5v without any logic level shifter for data pin) yields comparatively fewer occurrences(once in a 2day continuous test) of **_nan_** (DHT22 hang) issue

So the solution we came up with is 
1. Either powering down and reconnecting using a relay. 
2. or Powering the DHT22's VCC from a digital pin since it requires just a couple of mAmps and pulling down that pin for a sec when a **_nan_** (DHT22 hang) happens

> Sorry sounds like you've tried it all. The dht22 is highly timing dependent since it uses a single data pin that the mc and dht22 alternately control. I don't have the esp8266 or a dht22 to test on. [...]

Lordofrealms, 2019-07-16:
You could probably also put the data pin on an ADC pin and tweak the
library to do an analog read and shift the value of what it considers high
and low. Sounds like you're having issues with the voltage drop.

That is a lot of work rather than just throwing 5v on a GPIO pin though ;)

Glad you found a solution!

ahmadSum1, 2019-07-27:
@JorotoBG Yes, I have two versions of hardware, one with relay and another with powering the DHT's Vcc from a gpio.
Both solutions are working. In fact, one of the devices is installed on a remote location and didn't require maintenance for the last 10days.
This is what we did in the code:
```
Temperature = dht.readTemperature();
  delay(10);
  Humidity = dht.readHumidity();
  if (isnan(Temperature))
  {
    DHT_restart();
  }
```
and 
```
// Physically power recycle the DHT using a relay/GPIO
void DHT_restart() {
  digitalWrite(DHTVcc, LOW);
  digitalWrite(DHTPIN, LOW);
  delay(1000);
  digitalWrite(DHTVcc, HIGH);
  digitalWrite(DHTPIN, HIGH);
  delay(1000);
}
```

JorotoBG, 2019-07-27:
Dear ahmadSum1,  
Thank you for your reply.  Probably "isnan() "  will be and my solution (for now) . I read for the similar issues with DHT22(AM2302)  and AM2320 - when the humidity is around 60-80% -  "nan" is often happened.. 
Meanwhile I will continue with investigation of the problem and will share the results. 
Regards.

Tech-TX, 2019-10-08:
> I am having exactly the same problem with WEMOS Mini D1 and DHT22. I have found that the AESONG variant works fine and the ASAIR does not (using "Arduino" Pin 2 (D4) with/without pullup (10k builtin)..... hot plugging the AESONG one, after WEMOS initialisation starts it working.......

It's the same company, as far as I can see from their website.  They use the names interchangeably.  Maybe they made several 'bad' batches that have one or the other logo on them.

Here's a screencap from their website:
[image]

dmitriyteteruk, 2019-10-10:
Hi to all!  I am using version=1.3.7 of DHT sensor library and having same issue.
@ahmadSum1 thank you for your sketch with DHT reset, because in my case when all wires soldered I cannot to hard reset and can do only soft reset. I will try it.

> The code of V1.3.2 occasionnally misses the first LOW DHT signal of the 80us LOW/80us HIGH signal pair that preceeds the 40 bit information data stream. This is because occasionnally, the DHT takes [...]
@JmBelisle, can you specify where did you find DHT.cpp with code in lines 187, 174 and 182? I use orignal DHT.cpp from Adafriud and in this line I can see comments only :) So I do not know where to fix us40 to 55.

JmBelisle, 2019-10-11:
@DmiTryT

Version 1.3.7 by default introduces a wait of 55us (see default for parameter "usec" of the begin statement : line 50 in DHT.h).
This optional parameter to the "begin" statement can be provided to override the 55us default delay ... and set it to maybe more.
I recommend that you experiment with higher values and see if that improves your results.
For information I used Adafruit AM2302 units  (packaged DHT22 unit WITH 5.1K pullup resistor builtin between DATA and VCC).
I have had NO issues with NANs for last year in continuous use of several AM2302 connected to ESP8266 modules.

Be wary of versions 1.3.3 to 1.3.5 that had issues with side-effects of Interrupt locks around the critical 40 bit data capture code.
Version 1.3.7 works for me.
jmb

dmitriyteteruk, 2019-10-11:
@JmBelisle, Thank you for clarification! 
My solution in this case:
- I did change usec to 60 in DHT.h file from libriary.
```
class DHT {
  public:
   DHT(uint8_t pin, uint8_t type, uint8_t count=6);
   void begin(uint8_t usec=60); //55
```
- I added state change for GPIOs (OUT before read data and INPUT after read)
```
  pinMode(5, OUTPUT);
  float t1 = dht1.readTemperature();
  delay(2000);
  pinMode(5, INPUT);  

  pinMode(4, OUTPUT);
  float t2 = dht2.readTemperature();
  delay(2000);
  pinMode(4, INPUT); 
  
  pinMode(0, OUTPUT);
  float t3 = dht3.readTemperature();
  delay(2000);
  pinMode(0, INPUT); 
  
  pinMode(2, OUTPUT);  
  float t4 = dht4.readTemperature();
  delay(2000);
  pinMode(2, INPUT); 
```

- I added 5000 delay between dht begins as it is not critical to wait for me.
```
  dht1.begin();
  delay(5000);
  dht2.begin();
  delay(5000);
  dht3.begin();
  delay(5000);
  dht4.begin();
  delay(5000);
```
and now I can see data from all 4 sensors!!!

JmBelisle, 2019-10-11:
@DmitryT<mailto:notifications@github.com>
Good to hear you have solved your problem.
For your information, you did not have to modify the DHT.h file.
You could have also simply passed the new delay you wanted (60us) as a parameter to the "begin" function.
You could even have had different delays for different sensors.
such as:

  dht1.begin(60);
  delay(5000);
  dht2.begin(60);
  delay(5000);
  dht3.begin(75); // unit 3 is notorious in being a little slow (just an example)
  delay(5000);
  dht4.begin(60);
  delay(5000);

This way you do not have to modify the library code and it shields you from future updates to the library module.

Still, glad I could be of assistance.
Cheers
jmb

xtools-at, 2020-04-10:
all the suggestions here didn't help for my D1 Mini. Eventually went with library version 1.1.1. and everything works fine, just had to change `dht.readTemperature()` to `dht.readTemperature(false)`

Atyec89, 2020-04-19:
Hello,
Looks like it is not so new thread, but I want to share my experience.
After getting from AliExpress few Wemos D1 mini modules (ESP-12F), and few DHT22(AM2302) shields, I have got similar issue by getting NaN readings, when using libraries newer then 1.2.0.
After some hour/days of playing, a have got them work with modified 1.3.8 library.
After turning on debug messages, every time the following appeared:
"DHT timeout waiting for start signal high pulse." (line 289, DHT.cpp).
The problem behind this was, that after sending start signal from ESP, so pulling data line low, the delay was to short ->1100usec. According datasheet of sensor it should be enough, but 20msec is necessary in my case. So I simly have added an extra delay of 20msec after line 256. BTW, sometimes still getting NaN readings (1 out of 200-300). Maybe some fine tune is neccesary for this particular delay.

[image]

sinlac, 2020-04-26:
> Hello,
> Looks like it is not so new thread, but I want to share my experience.
> After getting from AliExpress few Wemos D1 mini modules (ESP-12F), and few DHT22(AM2302) shields, I have got similar issue by getting NaN readings, when using libraries newer then 1.2.0.
> After some hour/days of playing, a have got them work with modified 1.3.8 library.
> After turning on debug messages, every time the following appeared:
> "DHT timeout waiting for start signal high pulse." (line 289, DHT.cpp).
> The problem behind this was, that after sending start signal from ESP, so pulling data line low, the delay was to short ->1100usec. According datasheet of sensor it should be enough, but 20msec is necessary in my case. So I simply have added an extra delay of 20msec after line 256. BTW, sometimes still getting NaN readings (1 out of 200-300). Maybe some fine tune is necessary for this particular delay.
> 
> [image]

Two days I have been looking for this.
Thank you :)

NodeMCU V3 Lua

khjoen, 2020-04-30:
> Hello,
> Looks like it is not so new thread, but I want to share my experience.
> After getting from AliExpress few Wemos D1 mini modules (ESP-12F), and few DHT22(AM2302) shields, I have got similar [...]

Have you tried setting the type to DHT11 when instanciating the DHT object?

Atyec89, 2020-05-01:
> > Hello,
> > Looks like it is not so new thread, but I want to share my experience.
> > After getting from AliExpress few Wemos D1 mini modules (ESP-12F), and few DHT22(AM2302) shields, I have got similar issue by getting NaN readings, when using libraries newer then 1.2.0.
> > After some hour/days of playing, a have got them work with modified 1.3.8 library.
> > After turning on debug messages, every time the following appeared:
> > "DHT timeout waiting for start signal high pulse." (line 289, DHT.cpp).
> > The problem behind this was, that after sending start signal from ESP, so pulling data line low, the delay was to short ->1100usec. According datasheet of sensor it should be enough, but 20msec is necessary in my case. So I simply have added an extra delay of 20msec after line 256. BTW, sometimes still getting NaN readings (1 out of 200-300). Maybe some fine tune is necessary for this particular delay.
> > 
> > [image]
> 
> Have you tried setting the type to DHT11 when instanciating the DHT object?
> 
> Datasheet says between 1 and 10 ms delay for AM2302
> [image]
> 
> 

Yes I have. But then readings giving strange numbers. 
Nevertheless, a have opened the sensor, and under plastic something strange I found. Inside there is a sensor, but not am2302, but Sensirion SHTC3( with I2C interface). And some kind of microcontroller to simulate one wire protocol of dht22. I thin this is the reason of strange behavior...

mikekov, 2020-08-16:
Just my two cents... ESP32 Lolin32 board, Arduino framework, DHT library 1.3.10. Occasional NANs using DHT22 sensor.
As far as I could tell those were caused by no proper locking. Sensor `DHT::read()` uses `InterruptLock lock` which evaluates to `cli()`, which is defined as a no op in esp32/Arduino.h. The solution was to call `portDISABLE_INTERRUPTS()` before `readTemprature()`, follwed by `portENABLE_INTERRUPTS()`.

CONSULitAS, 2020-09-09:
## Just my 2 cents:

Trouble with the Wemos D1 mini DHT shield is a hardware bug!

On https://github.com/letscontrolit/ESPEasy/issues/2569#issuecomment-688901549 you will find output from logic analyzer measurements, detailed analysis what happens and a (hardware) fix. There is no software fix. 😢 

## Solution in short:

* D4 (GPIO2) does not work. 👎 
* DHT22 crashes by signals not understood while soft reset.
* Use other pin: D5 works fine.

ricardogomes1, 2020-09-09:
Boa tarde;

Sou bem iniciante no ramo e provavelmente encontrarao bastante erro no meu codigo porem estou tendo o mesmo problema porem apenas com valores abaixo de 0, temperaturas acima de 0 funciona perfeitamente e preciso medir temperaturas negativas.
Estou tentando utlizar a versao mais recente da biblioteca do dht22, segue o codigo:

`#include <WiFi.h>
#include <HTTPClient.h>
#include <WebServer.h>
#include "DHT.h"
//Ported to ESP32
#ifdef ESP32
#include <esp_wifi.h>
#include <WiFi.h>
#include <WiFiClient.h>
#define ESP_getChipId()   ((uint32_t)ESP.getEfuseMac())
#define LED_ON      HIGH
#define LED_OFF     LOW
#else
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#define ESP_getChipId()   (ESP.getChipId())
#define LED_ON      LOW
#define LED_OFF     HIGH
#endif
#include <ESP_WiFiManager.h>              //https://github.com/khoih-prog/ESP_WiFiManager
#include <MD5.h>

// SSID and PW for your Router
String Router_SSID;
String Router_Pass;

IPAddress stationIP   = IPAddress();
IPAddress gatewayIP   = IPAddress();
IPAddress netMask     = IPAddress();
IPAddress dns1IP      = gatewayIP;
IPAddress dns2IP      = IPAddress();

byte mac[6];
String macString = String (WiFi.macAddress());

void heartBeatPrint(void)
{
  static int num = 1;

  if (WiFi.status() == WL_CONNECTED)
    Serial.print("Conectado - ");        // C means connected to WiFi
  else
    Serial.print("Desconectado - ");        // D means not connected to WiFi

  if (num == 80)
  {
    Serial.println();
    num = 1;
  }
  else if (num++ % 10 == 0)
  {
    Serial.print(" ");
  }
}

void check_status()
{
  static ulong checkstatus_timeout = 0;

#define HEARTBEAT_INTERVAL    1000L
  // Print hearbeat every HEARTBEAT_INTERVAL (10) seconds.
  if ((millis() > checkstatus_timeout) || (checkstatus_timeout == 0))
  {
[... 151 lines cut ...]
      Serial.print(md5str);
      Serial.print("  ");

     stringPost = "{\"id\":\"" + macString + "\",\"reads\":\"" + temperatura + ";" + umidade + "\",\"token\":\"" + md5str + "\"}";

     https.begin("https://thermocontrol.teknisa.com/backend/insertRead"); //Especifique o destino para a solicitação HTTP
     https.addHeader("Content-Type","application/json");  //Especificar cabeçalho do tipo de conteúdo
 
     // Realiza o post para o endereço informado anteriormente
     int httpResponseCode = https.POST(stringPost); //Envie a solicitação POST real
     Serial.println (httpResponseCode);
    
     Serial.println(https.getString());

      //Somente para conferencia
     Serial.print("VALOR DO POST: ");  
     Serial.println(stringPost);

     //Give the Memory back to the System if you run the md5 Hash generation in a loop
     free(md5str);
}
else 
  ESP.restart();
  check_status();
}`

benhadad, 2021-02-19:
> I also had lot of issues with DHT22 and ESP8266-01. Used a 5.1k Pullup resistor. I was only getting readings from the sensor every 20th try, else just getting NAN.
> I also tried the code-changes suggested by [jaipharma](https://github.com/adafruit/DHT-sensor-library/issues/116#issuecomment-454261351), but that also didn't help.
> 
> I am using the same ESP8266-01 with DHT22 also with [nodemcu-firmware](https://github.com/nodemcu/nodemcu-firmware), and there it worked nicely. Therefore I have compared their [implementation ](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/dht/dht.c) (starting in line 250) with the one in this library and made few changes. Now it works nicely for me (haven't seen NAN since).
> 
> `
> uint32_t cycles[80];
> {
> // Turn off interrupts temporarily because the next sections
> // are timing critical and we don't want any interruptions.
> InterruptLock lock;
> 
> ```
> // End the start signal by setting data line high for 40 microseconds.
> digitalWrite(_pin, HIGH);
> delayMicroseconds(40);
> pinMode(_pin, INPUT);
> 
> if (expectPulse(LOW) == TIMEOUT) {
>   DEBUG_PRINTLN(F("DHT timeout waiting for start signal high pulse."));
>   _lastresult = false;
>   return _lastresult;
> }
> if (expectPulse(HIGH) == TIMEOUT) {
>   DEBUG_PRINTLN(F("DHT timeout waiting for start signal low pulse."));
>   _lastresult = false;
>   return _lastresult;
> } `
> ```
> 
> I think there is a error in the sensor data-sheet. I says, that the sensor will pull up bus's voltage within 20-40us, but I think the sensor expects the MCU to do that (also also shown by the color in the chart).
> 
> Hope this helps.

Is this the same solution as https://desire.giesecke.tk/index.php/2018/01/30/esp32-dht11/?

wlmqpsc, 2021-11-12:
Still have this problem on Node MCU 12E with 1.4.3
i connected them, add the 10K ohm pull up resistance. it can not get the number.
i have tried on two Node MCU boards and Arduino UNO. It only works on UNO.

LiaraAlis, 2022-01-22:
Same problem here. Only works with version 1.2.3. It doesn't work with versions from 1.3.x or above.

caternuson, 2022-12-06:
Closing. Seems fixed. Please only add additional commentary here if your are using an Adafruit ESP8266 based board.

DHT-sensor-library = 1.4.4
Arduino IDE = 1.8.19
ESP8266 BSP = 3.0.2
Adafruit Feather ESP8266 with DHT22 attached to pin 14 running `DHTtester.ino` sketch:
```
DHTxx test!
Humidity: 34.30%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.91°F
Humidity: 33.30%  Temperature: 19.90°C 67.82°F  Heat index: 18.81°C 65.87°F
Humidity: 34.50%  Temperature: 19.90°C 67.82°F  Heat index: 18.85°C 65.92°F
Humidity: 34.30%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.91°F
Humidity: 34.20%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.91°F
Humidity: 34.20%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.91°F
Humidity: 34.10%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.90°F
Humidity: 34.10%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.90°F
Humidity: 34.00%  Temperature: 19.90°C 67.82°F  Heat index: 18.83°C 65.90°F
Humidity: 34.00%  Temperature: 19.90°C 67.82°F  Heat index: 18.83°C 65.90°F
Humidity: 34.00%  Temperature: 19.90°C 67.82°F  Heat index: 18.83°C 65.90°F
Humidity: 34.00%  Temperature: 19.90°C 67.82°F  Heat index: 18.83°C 65.90°F
Humidity: 34.00%  Temperature: 19.90°C 67.82°F  Heat index: 18.83°C 65.90°F
Humidity: 34.00%  Temperature: 19.90°C 67.82°F  Heat index: 18.83°C 65.90°F
Humidity: 34.10%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.90°F
Humidity: 34.20%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.91°F
Humidity: 34.20%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.91°F
Humidity: 34.10%  Temperature: 19.90°C 67.82°F  Heat index: 18.84°C 65.90°F
```

yes-github, 2024-01-02:
> ## Just my 2 cents:
> Trouble with the Wemos D1 mini DHT shield is a hardware bug!
> 
> On [letscontrolit/ESPEasy#2569 (comment)](https://github.com/letscontrolit/ESPEasy/issues/2569#issuecomment-688901549) you will find output from logic analyzer measurements, detailed analysis what happens and a (hardware) fix. There is no software fix. 😢
> 
> ## Solution in short:
> * D4 (GPIO2) does not work. 👎
> * DHT22 crashes by signals not understood while soft reset. 💣
> * Use other pin: D5 works fine. 😲

[ESP8266 Pinout Reference: Which GPIO pins should you use?](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/)

nodeMCU esp8266 input recommended use ：D1/D2/D5/D6/D7/A0(Analog Input)