adafruit/DHT-sensor-library #121: RED LED DOESNT BLINK WHEN PLUGGED IN. MY PC CANNOT RECOGNIZED /FIND THE TRINKET 5V.

Poroppey, 2019-01-25:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

Poroppey, 2019-01-25:
I already upload the blink in basics and the pc recognize the trinket. then a few days ago, i didn't check or plugged the adafruit. and now the adafruit is not recognized in my pc. how can i trouble shoot this? one thing i did was i soldered it to secure in my breadboard dont tell me it is not good for soldering?

geofbaum, 2019-01-25:
@Poroppey Not sure why you are posting this an issue in the Adafruit DHT Sensor Library as your issue is clearly not with this sensor. Please only open an Issue if it has to do with this library and sensor. I'd suggest posting your problem on the Adafruit forum for the Trinket. Perhaps someone there can help you.

[https://forums.adafruit.com/viewforum.php?f=52&sid=7e03104a372fcb5a41d4020b506a6a32](https://forums.adafruit.com/viewforum.php?f=52&sid=7e03104a372fcb5a41d4020b506a6a32)

Poroppey, 2019-01-26:
I can say that I didnt do anything that made my trinket act like that. The
trinket is not being recognized by my laptop, but a few days ago it does.
Now the trinket is unknown, it says that it encountered problem with a code
(Code 43) i dont know why this happened. Honestly I dont know anything
about prograng this trinket but I have to do because of the research
project we are doing. I dont know what triggers that problem. ok

geofbaum, 2019-01-29:
@Poroppey Have you googled Arduino IDE Code 43? It seems like you need to reset your device which means something happened between the time when it was working and when it was not. Again the Adafruit and Arduino forums are a better place to ask about this and even more so it still has nothing to do with this code library or a DHT sensor. So this issue does not belong here and should be closed.

PaintYourDragon, 2019-02-14:
Closing issue since it’s not relevant. @Poroppey, please use forums.adafruit.com for general technical support, thanks. Github issues are for reporting bugs in specific code.

closed 2019-02-14 by PaintYourDragon