adafruit/DHT-sensor-library #134: Teensy 3.5 and DHT11

Lordofrealms, 2019-06-19:
DHT11 on D2 of TEENSY 3.5 powered with 5V or 3.3V not able to be read by this library or by idDHT11 library. Tried suggested fixes of changing pull-up delay time in declaration, but with no resolution.  Seems time scale related. idDHT11 library able to read for first few seconds then throws checksum errors continuously.

```
Retrieving information from sensor: Read sensor: OK
Humidity (%): 53.00
Temperature (oC): 24.00
Temperature (oF): 75.20
Temperature (K): 297.15
Dew Point (oC): 13.82
Dew Point Slow (oC): 13.85

Retrieving information from sensor: Read sensor: OK
Humidity (%): 54.00
Temperature (oC): 24.00
Temperature (oF): 75.20
Temperature (K): 297.15
Dew Point (oC): 14.11
Dew Point Slow (oC): 14.14

Retrieving information from sensor: Read sensor: OK
Humidity (%): 54.00
Temperature (oC): 24.00
Temperature (oF): 75.20
Temperature (K): 297.15
Dew Point (oC): 14.11
Dew Point Slow (oC): 14.14

Retrieving information from sensor: Read sensor: Error
Checksum error
Humidity (%): -1.00
Temperature (oC): -1.00
Temperature (oF): -1.00
Temperature (K): -1.00
Dew Point (oC): -1.00
Dew Point Slow (oC): -1.00
```

Lordofrealms, 2019-06-22:
This should be fixed on your library, but as an immediate alternative, I made a correct library here which should fix this issue as well as issues on esp MCs:
https://github.com/Lordofrealms/DHT11LUM

ladyada, 2019-07-20:
hi please submit PR's for fixes

caternuson, 2022-12-06:
Closing. Not finding any PRs submitted for this. There's been a lot of revisions to this library since issue was posted. Please update to latest releases. Can reopen if still an issue.

closed 2022-12-06 by caternuson