adafruit/DHT-sensor-library #147: Warning in library

Innominatam, 2020-01-03:
Hello!
I found a warning in your library in the "DHT.cpp" file.
"comparison between signed and unsigned integer expressions"
I think that it can be fixed by replacing on line 29
`#define TIMEOUT -1`
 with
`#define TIMEOUT 0xffffffffUL`
or maybe 
`#define TIMEOUT UINT32_MAX`

bugpics, 2020-05-14:
Hello,
You can add these warnings :

> DHT_sensor_library\DHT.cpp:40:45:` warning: unused parameter 'count' [-Wunused-parameter]  DHT::DHT(uint8_t` pin, uint8_t type, uint8_t count) {

> DHT_sensor_library\DHT.cpp:282:26: warning: comparison between signed and unsigned integer expressions [-Wsign-compare] if (expectPulse(LOW) == TIMEOUT) {

> DHT_sensor_library\DHT.cpp:287:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare] if (expectPulse(HIGH) == TIMEOUT) {

> DHT_sensor_library\DHT.cpp:312:20: warning: comparison between signed and unsigned integer expressions [-Wsign-compare] if ((lowCycles == TIMEOUT) || (highCycles == TIMEOUT)) {

> DHT_sensor_library\DHT.cpp:312:47: warning: comparison between signed and unsigned integer expressions [-Wsign-compare] if ((lowCycles == TIMEOUT) || (highCycles == TIMEOUT)) {

ladyada, 2020-05-14:
OK great, please submit PRs!

Rotzbua, 2020-05-21:
@ladyada I submitted 2 PRs to fix.

PR #160 (2020-05-21): Fix compiler warning: comparison between signed...
... and unsigned integer expressions [-Wsign-compare]

Fixes #147
closed 2020-05-21 by ladyada