adafruit/DHT-sensor-library #150: DHT11 isn't able to work with 3.3V

viktorzavadil, 2020-02-21:
Hi, I use your library for ESP01 and works perfect if I supply to DHT11 5V. But ESP01 works with 3.3V and I don't want use voltage level convertor and sensor should support 3.3V.

My issue is very similar to issue #98. I have tried to modify `DHT.cpp` by this comment: https://github.com/adafruit/DHT-sensor-library/issues/98#issuecomment-450399868 but it does not work.

Do you have some recommendation? Thank you very much.

ibebbs, 2020-02-26:
I am also encountering this issue. I am using a DHT11 with a (Arduino Compatible) Heltec CubeCell module which powers the DHT11 at 3.3v.

Interestingly while reading the DHT11 normally returns 'NaN', _occasionally_ it will actually read the device correctly. This leads me to believe there is likely to be a timing issue related to the lower power levels.

Be great if someone could provide advice as to how to get this working reliably. Thanks.

klavatron, 2020-03-20:
DHT11 works for my Esp8266-12f, 3.3v, 10k resistor, GPIO5. If trying to use A0 -> 'NaN'.

lestofante, 2021-01-04:
Same issue, i spoof the data PIN with a logic analyzer and the data is transmitted and received correctly.

data decoded from pulseview:
0x24 0x00 0x18 0x00 0x3c

data read by the lib:
0x92, 0x00, 0x0c, 0x00, 0x1E

lestofante, 2021-01-06:
update:
I did some debugging and found out the issue is very likely just timing (interrupt are disabled in theory, but maybe some task switching is going on).

As fix i rewrote the code to use a change interrupt; the following is my final code, tested with an DHT22 and and ESP32 and very similar code also tested with ESP8266 so it should work also there (end up the ESP8266 can have interrupt with pin in OUTPUT mode, but not the ESP32, so i had to move around the attachInterrupt)
Synchronization is ok as we have about 10uS to execute attachIntterupt before the chip should answer, and at the end of the communication there should be no data unless requested.

```
constexpr uint32_t ISR_ELEMENTS = 85;

struct {
  volatile uint32_t index;
  volatile uint32_t timigs[ISR_ELEMENTS];
} isr_data = {0};

void IRAM_ATTR ICACHE_RAM_ATTR read_dht(){
  if (isr_data.index < ISR_ELEMENTS) {
    isr_data.timigs[isr_data.index++] = micros();
  }
}

Reading read_sensor(){
  pinMode(DHT_PIN, OUTPUT);
  digitalWrite(DHT_PIN, HIGH);
  delay(1);
  
  // First set data line low for a period according to sensor type
  digitalWrite(DHT_PIN, LOW);
  delay(20); // data sheet says at least 18ms, 20ms just to be safe
  
  // End the start signal by setting data line high for 40 microseconds.
  pinMode(DHT_PIN, INPUT_PULLUP);

  attachInterrupt(digitalPinToInterrupt(DHT_PIN), read_dht, CHANGE);
  isr_data.index = 0;
  
  // Delay a moment to let sensor pull data line low.
  delayMicroseconds(10);
  
  // 3ms is the normal read time
  delay(4);

  if (isr_data.index != 84){
    return {millis() / 1000,0,0};
  }

  uint8_t data[5] = {0,0,0,0,0};
  uint8_t offset = 0;
  uint8_t shift = 7;
  for (uint32_t i = 4; i < isr_data.index; i+=2){
    bool bit = (isr_data.timigs[i] - isr_data.timigs[i-1]) > 50;
    data[offset] |= bit << shift;
    shift--;
    if (shift > 7){
      shift = 7;
      offset++;
    }
  }
```
Consider this code MIT license, as per the library itself

fixx-er, 2021-02-08:
Well it took me about 3 hours figguring this out.  I got so pissed off that I forgot to remove power to my D1-mini and the DH11.
I configured my D1-mini to use D8 and 3.3v pins for the DH11 data and power respectively.
- Attaching a 1k resistor between D8 and ground worked.
- Attaching a 10k resistor between D8 and ground did not work.
- Attaching a **5.1k resistor** between D8 and ground finally worked!
good luck and see ya all on the other side....

fixx-er, 2021-02-10:
> Well it took me about 3 hours figguring this out. I got so pissed off during my research that I forgot to remove power to my D1-mini and the DH11.
> I configured my D1-mini to use D8 and 3.3v pins for the DH11 data and power respectively.
> 
>     * Attaching a 1k resistor between D8 and ground worked.
> 
>     * Attaching a 10k resistor between D8 and ground did not work.
> 
>     * Attaching a **5.1k resistor** between D8 and ground finally worked!
>       good luck and see ya all on the other side....

I just found out that certain pins on the D!-mini and perhaps other ESP8266 chips, that certain pins will float low or high and for boot up conditions.  They must be pulled down or up to get past boot to function.  Below are my two sources.
https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/
https://escapequotes.net/esp8266-wemos-d1-mini-pins-and-diagram/

benhadad, 2021-02-19:
There is another library called  DHTx for ESP32 to handle interup issues.
https://desire.giesecke.tk/index.php/2018/01/30/esp32-dht11/
Not sure why both can't be merged.

caternuson, 2022-12-06:
Closing. Can't recreate. Possibly fixed by various updates to library, BSP, etc.

DHT-sensor-library = 1.4.4
Arduino IDE = 1.8.19
ESP8266 BSP = 3.0.2
Adafruit Feather ESP8266 with DHT11 attached to pin 14 and 3.3V power running `DHTtester.ino` sketch:
```
DHTxx test!
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 40.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.43°C 66.97°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 44.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.53°C 67.16°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
Humidity: 45.00%  Temperature: 20.30°C 68.54°F  Heat index: 19.56°C 67.21°F
```

closed 2022-12-06 by caternuson