adafruit/DHT-sensor-library #153: HeatIndex for lower values of temperature and humidity

frankielc, 2020-04-02:
The current algorithm used for calculating heat index takes into account Rothfusz equation to achieve the values that Robert G. Steadman came up by experimentation. It then has some refinements done by the NWS but doesn't consider the other refinements that NWS did for lower values of temperature and humidity. 

That's the reason why heat index for common house-hold values, like 74.84F and 68.5H give a heat index of 66.38F when in reality it should somewhere in the range of 75.23F - which makes more sense.

NWS full algo is very well detailed in this paper:
https://www.ncbi.nlm.nih.gov/pmc/articles/PMC3801457/

I will also submit a pull request.

MakerPact, 2026-10-27:
Has this been resolved? I’m trying to determine if additional work is still needed to implement this feature, or if the issue can simply be closed.

Daryavahus, 2026-10-27:
Thank You. It's working perfect.

frankielc, 2026-10-28:
**_Has this been resolved?_** 

On the current actual branch, no. Do take into account that the heat index is an abstraction that was created by Robert Steadman in the late 70s. It tried to measure something that pertains to human physiology and thus doesn't follow a precise mathematical scale. 

Soon after the National Weather Service adopted and modified his model. That was the one I submitted code for.

Two years after my pull request (https://github.com/adafruit/DHT-sensor-library/pull/154), Berkeley researchers updated the model for more aggressive climate conditions; which we're currently seeing more of. The study is published here: https://iopscience.iop.org/article/10.1088/1748-9326/ad3144

Now, my suggestion would be to simply check the NWS formula (https://www.wpc.ncep.noaa.gov/html/heatindex_equation.shtml) and vibe code it into DHT.cpp. That will give a massive improvement in values returned from DHT.

MakerPact, 2026-10-29:
Im happy to do this.