adafruit/DHT-sensor-library #164: Delay in readings of DHT22

tormes, 2020-06-07:
In the file **DHT_U.cpp** , in function **void DHT_Unified::setMinDelay(sensor_t *sensor)** , for the sensors DHT21, DHT22, the **sensor->min_delay** should be 0.5 sec, not 2 secs.
This sensors (they are better than the DHT11) appears to have a maximum frequency of 2 Hz = 2 times a second, so the minimum time between readings should be 0.5 secs.

RichardSHutchison, 2022-04-29:
It's been 2 years and so you may have already figured this out. If you look around at the DHT22 specs, the sample rate is 0.5Hz. That stand's for half a reading per second, or one reading every two seconds. So the min_delay value of 2 seconds is actually correct.

The DHT22 has better temperature and humidity ranges and tolerances than the DHT11 but its sampling rate is slower.

Check this out: https://learn.adafruit.com/dht
Or the spec sheet: https://www.electroschematics.com/wp-content/uploads/2015/02/DHT22-datasheet.pdf

caternuson, 2022-12-06:
Yep. 2 seconds. From DHT22 datasheet linked above:
[image]

closed 2022-12-06 by caternuson