adafruit/DHT-sensor-library #166: New feature suggestion - online Simulation page for DHT library

Puneethkumarsn, 2020-08-26:
hi there

Thank you so much for the example library which is well written for the DHT sensor. It is very helpful for us and as well as many others using the sensor for the first time.

We also have been working on the DHT22 simulator at https://wokwi.com/ which can run the DHT library example without leaving the browser. you can find the preview window here. It is free and will be helpful to the users of the library to get a preview of the library online before they do establish the actual connection between Arduino and the DHT sensor.

I hope this is interesting and useful. Please let me know if I can share a link to the simulation page for the example library. I can create a PR to add the links to the simulation pages, which the user will be able to access.
Kindly share your feedback on the simulation page as well. if there ate nay improvements or changes to be done. We will be happy to add it as well. 

Example link: https://wokwi.com/arduino/libraries/DHTlib/dht22_test

[image]

ladyada, 2020-08-27:
I think we would add a note here
https://learn.adafruit.com/dht/downloads
in the tutorial
OK?

Puneethkumarsn, 2020-08-28:
Yes. We will be happy to provide direct links to the simulated page. Shall we share the link? The users can view the library in action with just one click. They will also be able to paly around with the code still.

Thank you for your feedback 😀😀

Puneethkumarsn, 2020-08-28:
Hi there,

We are sorry for the delay. here is the link which presnts the user both library examples

https://wokwi.com/arduino/libraries/DHT-sensor-library

Kindly leave a comment or a message once you add it to the downloads section. We will be thrilled to see.
-------------------------------------------------------------
It so happens that we also have OLED examples for Adafruit here: if you are interested we can do more :) 
https://wokwi.com/arduino/libraries/Adafruit_SSD1306

Kindly let us know

ladyada, 2020-08-30:
thanks we added it!

closed 2020-08-30 by ladyada

Puneethkumarsn, 2020-09-01:
Hi There,

there is a minor correction needed: 

https://wokwi.com/arduino/libraries/DHT-sensor-library is the correct link for the Adafruit library. I request you to kindly update so that, the link points to the Adafruit Library.

Many thanks 😀

ladyada, 2020-09-01:
fixed

Puneethkumarsn, 2020-09-01:
Many Thanks 😀😀