adafruit/DHT-sensor-library #172: Compile error using DHT library 1.4.0 on Arduino Uno Wifi Rev 2

MichelRabozee, 2020-11-19:
Hello, I updated my DHT library to version 1.4.0. I use an Arduino Uno WiFi Rev 2. I had no issue with DHT library version 1.3.10. Now, I get:
``
Arduino: 1.8.13 (Mac OS X), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

In file included from /Volumes/Michel_Alu_Sources/Arduino/libraries/DHT_sensor_library/DHT.cpp:26:0:
/Volumes/Michel_Alu_Sources/Arduino/libraries/DHT_sensor_library/DHT.cpp: In constructor 'DHT::DHT(uint8_t, uint8_t, uint8_t)':
/Volumes/Michel_Alu_Sources/Arduino/libraries/DHT_sensor_library/DHT.h:56:46: error: 'SystemCoreClock' was not declared in this scope
 #define microsecondsToClockCycles(a) ((a) * (SystemCoreClock / 1000000L))
                                              ^
/Volumes/Michel_Alu_Sources/Arduino/libraries/DHT_sensor_library/DHT.cpp:51:7: note: in expansion of macro 'microsecondsToClockCycles'
       microsecondsToClockCycles(1000); // 1 millisecond timeout for
       ^~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Uno WiFi Rev2.
``

The only differences between 1.3.10 and 1.4.0 is in DHT.h:
``
#if (TARGET_NAME == ARDUINO_NANO33BLE)
#ifndef microsecondsToClockCycles
/*!
 * As of 7 Sep 2020 the Arduino Nano 33 BLE boards do not have
 * microsecondsToClockCycles defined.
 */
#define microsecondsToClockCycles(a) ((a) * (SystemCoreClock / 1000000L))
#endif
#endif
``

How is it possible that the Arduino Uno WiFi Rev 2 defines ARDUINO_NANO33BLE ???

scottcowen, 2020-12-01:
On the nano every it states the systemCoreClock is not defined

ladyada, 2020-12-02:
@per1234 hi please ask someone with nano33 BLE / Wifi Rev 2 experience to take a look

per1234, 2020-12-02:
Hello! The reason this is evaluating to `true` is because neither of those identifiers are defined. From the C++11 standard sec. 16.1, paragraph 4:
>After all replacements due to macro expansion and the defined unary operator
have been performed, all remaining identifiers and keywords 148 , except for true and false, are replaced
with the pp-number 0

So to the C++ preprocessor this line:
```c++
#if (TARGET_NAME == ARDUINO_NANO33BLE)
```
Is the equivalent of:
```c++
#if (0 == 0)
```
You could fix the problem by changing it to something like this:
```c++
#if defined(TARGET_NAME) && (TARGET_NAME == ARDUINO_NANO33BLE)
```

ladyada, 2020-12-02:
kk if someone can submit a PR ill review it, i do not own the hardware to test against.

per1234, 2020-12-02:
OK, you got it: https://github.com/adafruit/DHT-sensor-library/pull/174

closed 2020-12-02 by ladyada