adafruit/DHT-sensor-library #177: Problem: wrong values at negative temperatures an ESP8266 and DHT22

argltuc, 2021-02-01:
- Arduino board:  ESP8266 / NodeMCU 1.0 (ESP12-E Module)
- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13
- installed DHT-sensor-library version: 1.4.1 / installed by Arduino IDE Lib manager

I use the DHT22 in combination with an ESP8266 and this lib.
When having negative temperatures, the resulting value of readTemperature() is -3278.6 at -0.1°C
To fix negatives values a recalculation is nessecary:

```
#include <stdint.h>
float temperature = dht.readTemperature();
if(temperature < 0)
 {
   temperature = (temperature = (INT16_MAX - temperature*-10)*-0.1; - temperature*-10)*-0.1;
 }
 ```
 
after some research, i figure out, that the values in data-array repressents an intager in ones complement, while the rest of data handling seems to be in BCD.
I will provide an PR to fix this issue.

tranquil-c, 2023-04-11:
I have been seeing this issue and have yet to narrow down on the cause.  

I'm using version 1.4.4 of this library and building with PlatformIO: 
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7)
 - tool-esptoolpy @ 1.40500.0 (4.5.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
 
I've observed it twice now.  On both occasions the temperature decreased, crossing 0 degrees, over the course of roughly an hour. 

I'll be investigating the issue and will provide an update should I find anything notable.

naymore, 2023-12-03:
I know this thread is quite a bit old but I couldn't find any real solution to the problem.
I'm using v1.4.6 of this library (on an ESP8266) and I  can confirm the problem is real.