adafruit/DHT-sensor-library #180: DHT11 with Feather 32u4 Lora - Can't read DHT

chrisShick, 2021-02-11:
- Arduino board:  Feather 32u4 LoRa

- Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.13

I followed the adafruit tutorial on setting up a DHT11 (3 pin) with the feather. The library just gives nan for both. However, when I run the library with an UNO it works just fine. Any ideas here? My thoughts are around the underlying timing issues.

roelvandun, 2021-03-14:
I have a similar problem, I also get a NaN for both on my Uno. But if I downgrade the library from 1.4.1 to 1.4.0, it magically works

caternuson, 2022-12-06:
Can not recreate the issue. Running the `DHTtest.ino` sketch on a Feather 32u4 with DHT11 on pin 3 runs as expected:
```
DHTxx test!
Humidity: 46.00%  Temperature: 19.70°C 67.46°F  Heat index: 18.93°C 66.07°F
Humidity: 42.00%  Temperature: 19.50°C 67.10°F  Heat index: 18.60°C 65.48°F
Humidity: 47.00%  Temperature: 19.50°C 67.10°F  Heat index: 18.73°C 65.72°F
Humidity: 47.00%  Temperature: 19.50°C 67.10°F  Heat index: 18.73°C 65.72°F
Humidity: 47.00%  Temperature: 19.60°C 67.28°F  Heat index: 18.84°C 65.92°F
Humidity: 47.00%  Temperature: 19.60°C 67.28°F  Heat index: 18.84°C 65.92°F
```

If this is still not working with your setup, please post in the forums with photos showing how everything is connected.
https://forums.adafruit.com/

closed 2022-12-06 by caternuson