adafruit/DHT-sensor-library #181: DHT11 wont read below 0 degree celsius

beicnet, 2021-02-11:
Hi there,

I'm using Arduino UNO and the new variant of the Asair DHT11 (not the Aosong) sensor over 3.3v.

[image]

Everything is working just fine until the sensor reading get to 0 degree celsius and it just wont go below.

Outside right now is -3c

btw. I'm using library version 1.3.0

Thank you for your support!

dl2sba, 2021-02-16:
I have fixed this code in DHT.cpp. 

My sensor also has the "feature" that it reports 0x0000 for 0°C as well as 0x8000 for 0°C.
Following the logic this is -327.7°C ...

```
 case DHT22:
    case DHT21:
      int16_t d = ((int16_t)data[2]) << 8 | data[3];
	  //	msb set == negativ
	  if (d & 0x8000) {
		  d &= 0x7fff;
		  //	0x8000 == 0x0000 == 0)
		  if ( d == 0 ) d = 0x7fff;
		  d = 0x7fff - d;
		  d *= -1;
	  }
	  f = d * 0.1f;

      if (S) {
        f = convertCtoF(f);
      }
      break;
```

Daryavahus, 2025-02-11:
It won't work. DHT Asair shows abnormal temperatures above 0 degree. Instead 23 it show 260 degree.

MaderDash, 2025-05-05:
Has this issue been resolved at this time? If so, it can be closed; if not, please let me know.

Daryavahus, 2025-05-05:
I tried many possibilities. And now my code looks like this:
header:
#define ADJUST 3
.....
		  	f = (data[2] & 0xFF) + ((data[3] & 0x7F) * 0.1F);
			if(data[3] & 0x80){
				f = -1.F - f;
			}
			  f -= ADJUST; // temp. correction
	 if (S) {
        f = convertCtoF(f);
      }