adafruit/DHT-sensor-library #182: Problem: DHT22 does not send data in sign-magnitude format

argltuc, 2021-02-12:
- Arduino board 1: ESP8266 / NodeMCU 1.0 (ESP12-E Module)
- Arduino board 2: ESP-WROOM-32 / ESP32 DEVKIT V1
- Arduino IDE version (found in Arduino -> About Arduino menu): 1.8.13
- installed DHT-sensor-library version: 1.4.1 / installed by Arduino IDE Lib manager

This problem is directly connected to #177

After some more research on negative temperatures with DHT22 and ESP8266 / ESP32, i figure out, that my DHT22 sensor does not send data as expected.

This library assumed, sensor temperature data is in sign-magnitude format. This is also documented here: https://cdn-shop.adafruit.com/datasheets/Digital+humidity+and+temperature+sensor+AM2302.pdf

But this DHT22 which I have available for testing, seems to send data in two's complement format. With one exception:
During transition from positive to negative values, there is one point with sign-magnitude.
Going from 0.1°C (0x0001) -> 0°C (0x0000) -> -0°C (0x8000) -> -0.1°C (0xFFFF)

I have also a log of this, where I add a hex print of data[2]/data[3] at end of readTemperature():
```
22:07:01.473 -> 0003    0.3
22:07:01.473 -> Temperature: 0.30C
22:07:03.485 -> 8000    -3276.8
22:07:03.485 -> Temperature: -3276.80C
22:07:05.500 -> FFFD    -0.3
22:07:05.500 -> Temperature: -0.30C
```

dl2sba, 2021-02-17:
Also my finding - I have fixed this code in DHT.cpp.

 ```
    case DHT22:
    case DHT21:
      int16_t d = ((int16_t)data[2]) << 8 | data[3];
	  //	msb set == negativ
	  if (d & 0x8000) {
		  d &= 0x7fff;
		  //	0x8000 == 0x0000 == 0)
		  if ( d == 0 ) d = 0x7fff;
		  d = 0x7fff - d;
		  d *= -1;
	  }
	  f = d * 0.1f;

      if (S) {
        f = convertCtoF(f);
      }
      break;
```

JavanXD, 2021-03-06:
Having the same problem on Raspberry Pi using the latest Adafruit_DHT package.

RamitArko, 2023-02-26:
I am also facing the same problem using Raspberry Pi 4