adafruit/DHT-sensor-library #193: ESP32 "Failed to read from DHT sensor!"

andi07x, 2022-07-30:
I was also struggling with the "Failed to read from DHT sensor!" message, with my ESP32 CAM Board. But after some googling I found something. On the page: https://github.com/mkjanke/ESP32-C3-Fanspeed I found out that the time critical code for the ESP32 is not protected. After changing the DHT.h file as indicated on the page, the values were reliably delivered. May be someone can add this here. Thank you.

1technophile, 2022-10-05:
Hello,

Having the same [issue ](https://community.openmqttgateway.com/t/dht22-sensor-not-showing/2045/22) with [OpenMQTTGateway ](https://docs.openmqttgateway.com) also 

The more up-to-date combination of libraries that works with ESP32 and DHT22 is the following:
```
dht = DHT sensor library@1.3.2
unifiedsensor = Adafruit Unified Sensor@1.1.4
```

It fails with:
```
dht = DHT sensor library@1.3.3
unifiedsensor = Adafruit Unified Sensor@1.1.4
```

caternuson, 2022-12-06:
Testing `DHT_Unified_Sensor.ino` example on a Qt PY ESP32 with DHT22 attached to pin A0, it works as expected:
```
DHTxx Unified Sensor Example
------------------------------------
Temperature Sensor
Sensor Type: DHT22
Driver Ver:  1
Unique ID:   -1
Max Value:   125.00°C
Min Value:   -40.00°C
Resolution:  0.10°C
------------------------------------
Humidity Sensor
Sensor Type: DHT22
Driver Ver:  1
Unique ID:   -1
Max Value:   100.00%
Min Value:   0.00%
Resolution:  0.10%
------------------------------------
Temperature: 19.80°C
Humidity: 34.90%
Temperature: 19.80°C
Humidity: 33.90%
Temperature: 19.80°C
Humidity: 35.00%
Temperature: 19.80°C
Humidity: 34.80%
Temperature: 19.80°C
Humidity: 34.60%
Temperature: 19.80°C
Humidity: 34.40%
Temperature: 19.80°C
Humidity: 34.30%
Temperature: 19.80°C
Humidity: 34.20%
Temperature: 19.90°C
Humidity: 34.20%
Temperature: 19.80°C
Humidity: 34.10%
Temperature: 19.80°C
Humidity: 34.10%
Temperature: 19.80°C
Humidity: 34.10%
```

Not sure what exactly original issue is referring to. Closing for now. Can reopen if still an issue - **please provide the most simple complete sketch that can be used to demonstrate the issue for testing.**

closed 2022-12-06 by caternuson