adafruit/DHT-sensor-library #202: Unable to read data on esp32c3

flex380, 2023-04-05:
Failed to read from DHT sensor!

phybros, 2023-04-12:
I have the same issue with an ESP32-C3-WROOM-02 module. I did find this person's repo (https://github.com/mkjanke/ESP32-C3-Fanspeed) which mentions that a change to the `InterruptLock` class is required to make the DHT readings work, but I haven't tried it:

```
class InterruptLock {
#if defined(ESP32)
  portMUX_TYPE mutex = portMUX_INITIALIZER_UNLOCKED;
#endif

public:
  InterruptLock() {
#if !defined(ARDUINO_ARCH_NRF52)
    noInterrupts();
#endif
#if defined(ESP32)
    portENTER_CRITICAL(&mutex);
#endif
  }
  ~InterruptLock() {
#if !defined(ARDUINO_ARCH_NRF52)
    interrupts();
#endif
#if defined(ESP32)
    portEXIT_CRITICAL(&mutex);
#endif

  }
};
```

Maybe we could just PR this to the library?

phybros, 2023-04-25:
I tried this ^^^ and it just makes the esp crash and dump