adafruit/DHT-sensor-library #206: Incorrect frequency limit for DHT11

cryptoAlgorithm, 2023-09-04:
- Arduino board: Raspberry Pi Pico (RP2040 B1)

- Arduino IDE version (found in Arduino -> About Arduino menu): PlatformIO `earlephilhower` core

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):

1. Read values from the sensor in a loop: `dht.readHumidity()` or ` dht.readTemperature()`
2. Observe that values do not change at a frequency higher than 0.5Hz

It appears that this is due to an artificial ratelimit imposed by the `DHT::read(...)` method. Specifically, `MIN_INTERVAL` is hardcoded to `2000` for all types of sensors, which I assume is due to the DHT22's maximum read frequency of 0.5Hz. However, other sensors supported by this library, such as the DHT11, can read at a higher rate of 1Hz, so this value should instead be set based on the sensor type.

MaderDash, 2025-05-05:
It appears that this issue is currently resolved. If it is not, please inform me. Otherwise, the ticket can be closed.

[image]
This is my reference @cryptoAlgorithm

cryptoAlgorithm, 2025-05-05:
@MaderDash Nope, the file in question is `DHT.cpp`, not the "unified library" in `DHT_U.cpp`. 

https://github.com/adafruit/DHT-sensor-library/blob/2295fe471c38d5e649a7b68cecccc42193c8e41c/DHT.cpp#L28

In the vanilla DHT library in this repository, the minimum interval is hardcoded with a preprocessor definition to be 2000ms (0.5Hz).