adafruit/DHT-sensor-library #211: Core  1 panic'ed (Interrupt wdt timeout on CPU1) on ESP32 NodeMCU ( ESP32-D0WD-V3)

mathieucarbou, 2024-01-07:
code:

called from stup():

```c++
    _dht = new DHT_Unified(_pin, _type == TemperatureSensorType::DHT11 ? DHT11 : DHT22);
    _dht->begin();
```

called from loop, executed each 10 seconds:

```c++
      sensors_event_t eventTemp;
      _dht->temperature().getEvent(&eventTemp);
      read = eventTemp.temperature;
```

crash:

```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4012963b  PS      : 0x00060235  A0      : 0x80127795  A1      : 0x3ffd4b80  
A2      : 0x00000004  A3      : 0x00000003  A4      : 0x00000027  A5      : 0x3ffef0f0  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x3ff44000  A9      : 0x0000001f  
A10     : 0x3ff5f07c  A11     : 0x80000001  A12     : 0x80091770  A13     : 0x3ffed800  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40084859  LEND    : 0x40084861  LCOUNT  : 0x00000027  

Backtrace: 0x40129638:0x3ffd4b80 0x40127792:0x3ffd4bb0 0x401b63e9:0x3ffd4bd0 0x401b64e2:0x3ffd4bf0 0x401b65c3:0x3ffd4d50 0x401b60a9:0x3ffd4d80 0x40114671:0x3ffd4da0 0x40114849:0x3ffd4e30 0x400f4e3a:0x3ffd4e50 0x40128aa1:0x3ffd4e70

  #0  0x40129638:0x3ffd4b80 in gpio_ll_get_level at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/gpio_ll.h:473
      (inlined by) gpio_get_level at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/gpio.c:233
  #1  0x40127792:0x3ffd4bb0 in __digitalRead at /Users/mat/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.c:150
  #2  0x401b63e9:0x3ffd4bd0 in DHT::expectPulse(bool) at .pio/libdeps/pro-esp32-debug/DHT sensor library/DHT.cpp:382
  #3  0x401b64e2:0x3ffd4bf0 in DHT::read(bool) at .pio/libdeps/pro-esp32-debug/DHT sensor library/DHT.cpp:306
  #4  0x401b65c3:0x3ffd4d50 in DHT::readTemperature(bool, bool) at .pio/libdeps/pro-esp32-debug/DHT sensor library/DHT.cpp:88
  #5  0x401b60a9:0x3ffd4d80 in DHT_Unified::Temperature::getEvent(sensors_event_t*) at .pio/libdeps/pro-esp32-debug/DHT sensor library/DHT_U.cpp:118
  #6  0x40114671:0x3ffd4da0 in Mycila::TemperatureSensor::_read() at lib/MycilaTemperatureSensor/MycilaTemperatureSensor.cpp:91
  #7  0x40114849:0x3ffd4e30 in Mycila::TemperatureSensor::loop() at lib/MycilaTemperatureSensor/MycilaTemperatureSensor.cpp:81
  #8  0x400f4e3a:0x3ffd4e50 in loop() at src/main.cpp:140
  #9  0x40128aa1:0x3ffd4e70 in loopTask(void*) at /Users/mat/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

Core  0 register dump:
PC      : 0x401da36a  PS      : 0x00060435  A0      : 0x80132e6c  A1      : 0x3ffbd5e0  
A2      : 0x00000000  A3      : 0x80000001  A4      : 0x80091770  A5      : 0x3ffe1db0  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x8013294e  A9      : 0x3ffbd5b0  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x80091770  A13     : 0x3ffbd4e0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x401da367:0x3ffbd5e0 0x40132e69:0x3ffbd600 0x4008f76c:0x3ffbd620

  #0  0x401da367:0x3ffbd5e0 in cpu_ll_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/cpu_ll.h:183
      (inlined by) esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:830
  #1  0x40132e69:0x3ffbd600 in esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/freertos_hooks.c:63
  #2  0x4008f76c:0x3ffbd620 in prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3987

ELF file SHA256: 6de87c414e8eed59

E (2638) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```

where it crashed:

```c++
// Otherwise fall back to using digitalRead (this seems to be necessary on
// ESP8266 right now, perhaps bugs in direct port access functions?).
#else
  while (digitalRead(_pin) == level) {
    if (count++ >= _maxcycles) {
      return TIMEOUT; // Exceeded timeout, fail.
    }
  }
#endif
```

Please let me know if you need something else...

mathieucarbou, 2024-01-07:
It seems that the watchdog triggers because of the loop.

I am trying for now to run with the watchdog disabled as indicated here and will report back in a few hours: https://stackoverflow.com/questions/51750377/how-to-disable-interrupt-watchdog-in-esp32-or-increase-isr-time-limit

mathieucarbou, 2024-01-07:
Might be related to https://github.com/adafruit/DHT-sensor-library/pull/93

jachik, 2024-12-10:
I have the same issue wie an ESP32 and a DHT22 sensor, but sometimes it takes some days after the error occurs. I think the problem is, that the loop in line 304 is missing a break condition. If we are getting a timeout for every expectPulse it takes about 80 ms and this is not far away from 300ms what is the watchdog timer. So something like this could help:

`
      if (cycles[i]==TIMEOUT || cycles[i+1]==TIMEOUT){
        DEBUG_PRINTLN(F("DHT timeout waiting for data signal."));
        _lastresult = false;
        return _lastresult;
      }
` 

I'm not able to test this right now, but on weekend I will test it and report back.

mathieucarbou, 2024-12-10:
I solved that by doing my own lib (https://github.com/mathieucarbou/MycilaDS18) which is using behind the code from https://github.com/junkfix/esp32-ds18b20 based on RMT peripheral. It is far more reliable on ESP32 and does not have bisy loops for the readings.

jachik, 2024-12-19:
As mentioned in (https://github.com/adafruit/DHT-sensor-library/issues/211#issuecomment-2530576418), I was able to test the modification and now for about 8 days my project is running without any errors. I have created a pull request (https://github.com/adafruit/DHT-sensor-library/pull/218).