adafruit/DHT-sensor-library #221: [DHT22] Incorrect handling of negative temperature values

bkormann80, 2024-12-29:
The current release 1.4.6 of DHT-sensor-library supports various DHTxx sensors. I tested several DHT22 sensors on ESP8266 and Arduino Uno experiencing correct behavior for humidity and positive temperature values. As soon as the temperature drops below zero, incorrect negative values are determined by `float DHT::readTemperature(bool S, bool force)` function in `DHT.cpp`. 

The data sheet of [(DHT22)](https://cdn-shop.adafruit.com/datasheets/Digital%20humidity%20and%20temperature%20sensor%20AM2302.pdf) explains the temperature data extraction on page 3. The MCU receives 40 bits (5 * 8 bits) with the following structure:

- 1st 16 bits: RH data (relative humidity)
- 2nd 16 bits: T data (temperature)
- last 8 bits: check sum

The current implementation follows the description of the data sheet, but returns incorrect data. A detailed inspection revealed, that temperature data is returned as a 16 bit signed integer value, hence no distinction between negative or positive values is needed. The `float DHT::readTemperature(bool S, bool force)` function in `DHT.cpp` must be adapted to:

```
float DHT::readTemperature(bool S, bool force) {
  float f = NAN;
  short s = 0;
...
    case DHT22:
      s = ((short) data[2]) << 8 | data[3];
      f = s * 0.1;
      if (S) {
        f = convertCtoF(f);
      }
      break;
...
```

The above mentioned patch solves the previous issue of incorrect negative temperature values. It seems like an inconsistency between documentation (data sheet) and sensor implementation or it might be a sensor variant to be supported by the DHT-sensor-library.

dhalbert, 2024-12-29:
Thanks for the diagnosis! Would you be willing to submit a pull request?

bkormann80, 2024-12-29:
Sure. I already added the changes to a branch of the DHT-sensor-library repo locally.

haas26p-ctrl, 2026-09-06:
Confirming this is still relevant for DHT22 below 0 °C, and that open PR #223 already implements the signed 16-bit interpretation from the issue (`short` from `data[2]/data[3]`, then `* 0.1`).

I am not opening a competing PR. Maintainers: #223 looks like the right fix to land (after rebase if needed — GitHub currently marks it unmergeable). Happy to help rebaseline that branch if useful.